thomas-krenn/check_ipmi_sensor_v3 #3: Hide events from event type "Entity Absent"

stephanhendl, 2016-04-25:
It would be great if one can hide events from the type "Entity Absent" since our Fujitsu servers show critical events from FANs that are not present. I would prefer a more generic solution rather than deselecting the IPMI IDs. 

```
ID  | Name             | Type                        | State    | Reading    | Units | Event 
117 | FAN9 SYS         | Entity Presence             | Critical | N/A        | N/A   | 'Entity Absent'
118 | FAN10 SYS        | Entity Presence             | Critical | N/A        | N/A   | 'Entity Absent'
119 | FAN11 SYS        | Entity Presence             | Critical | N/A        | N/A   | 'Entity Absent'
120 | FAN12 SYS        | Entity Presence             | Critical | N/A        | N/A   | 'Entity Absent'
121 | FAN13 SYS        | Entity Presence             | Critical | N/A        | N/A   | 'Entity Absent'
122 | FAN14 SYS        | Entity Presence             | Critical | N/A        | N/A   | 'Entity Absent'
123 | FAN15 SYS        | Entity Presence             | Critical | N/A        | N/A   | 'Entity Absent'
124 | FAN16 SYS        | Entity Presence             | Critical | N/A        | N/A   | 'Entity Absent'
131 | LOM              | Entity Presence             | Critical | N/A        | N/A   | 'Entity Absent'
133 | LOM              | Entity Presence             | Critical | N/A        | N/A   | 'Entity Absent'
```

Regards,
Stephan

check_ipmi_version: 3.10
ipmi-sensors: 1.4.5

veitw, 2016-04-25:
Hi,

I experience this with IBM/Lenovo machines, too, also for optional backplanes that are not installed. But I advocate to retain on the current behaviour, or at least make ignoring "entity absent" entries only an optional, non-default behaviour:

I strongly prefer enumerating IDs of optional devices that are not installed once over ignoring them as automatically ignoring "entity absent" events might not trigger monitoring errors in situations where a component has been removed or not properly replaced.
When the system is currently completely unpowered (as it is common for replacing cold-swap components) while a component becomes finally or flaky defective or is improperly replaced, the BMC might even miss to add a "component removed" event to the SEL.

In such a situation, only a previously created list of absent devices representing the system's regular state of device presence guarantees instant detection of hardware being removed or appearing absent due to defects or improper installation.

Best regards,
// Veit

stephanhendl, 2016-04-25:
Well, I agree with Veit that it should be NOT the default but maybe a new switch like "--no-entity-absent" could be a solution.

Best regards, Stephan

gschoenberger, 2016-05-14:
Can you try latest master with commit c4d7150d87337c3d178cae7de5f12a2d1dab7c9a?
Use '--noentityabsent' as option,

THX, Georg

gschoenberger, 2016-05-19:
I think I got it, some quotes were missing: 84245972ee4a37c4697776b743621d93b452b52f
Give it a try?

gschoenberger, 2016-05-24:
Closing issue as 8424597 fixed it.

closed 2016-05-24 by gschoenberger