thomas-krenn/check_ipmi_sensor_v3 #4: Unrecognized options for ipmi-sel

lckarssen, 2016-05-25:
I am trying to run this check on Ubuntu 14.04 and get the following error when running `./check_ipmi_sensor -vvv`:
```
/usr/sbin/ipmi-sel: unrecognized option '--output-event-state'
Try `ipmi-sel --help' or `ipmi-sel --usage' for more information.

-> Execution of /usr/sbin/ipmi-sel failed with return code 64.
-> /usr/sbin/ipmi-sel was executed with the following parameters:
   sudo /usr/sbin/ipmi-sel --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all
```

Running the command on the last line by hand I found out that this version of `ipmi-sel` doesn't support the following options: `--output-event-state` and `--sensor-types=all`. Removing these option from the command line at least makes the `ipmi-sel` command run. I haven't yet tested what the effect of removing these options is on the total check.

The `freeipmi-tools` package has version 0.8.12 in the Ubuntu 14.04 LTS repositories. On Ubuntu 16.04 the check works fine, there the package version is 1.4.11. Maybe the minimally required version of FreeIPMI should be increased in the README?

tk-wfischer, 2016-05-25:
It seems that you have a wrong (old) version of the package.

In the Ubuntu 10.04 LTS repository is version 1.1.5, see:
http://packages.ubuntu.com/trusty/freeipmi-tools

In case you are using 12.04 LTS you can use the Thomas-Krenn repository to get a newer version than 0.8.12 of FreeIPMI:
https://www.thomas-krenn.com/en/wiki/Thomas_Krenn_Ubuntu_Repository

Please let me know if this works, than I would close this issue.

Best regards,
Werner

lckarssen, 2016-05-25:
I'm sorry, you are right about the Ubuntu version. This was on a 12.04 LTS machinen not 14.04. 
Thank you for pointing to the repository. I will give it a try.

tk-wfischer, 2016-05-25:
No problem :-)
Our repository should work. I'll close this issue. In case that you experience any errors, please re-open it.

closed 2016-05-25 by tk-wfischer