thomas-krenn/check_ipmi_sensor_v3 #5: Upgrade to nagios-plugins-thomas-krenn 1.7-2 not working

localguru, 2016-05-27:
Hi Werner,

I've upgraded nagios-plugins-thomas-krenn 1.6-1 to
nagios-plugins-thomas-krenn all 1.7-2 on trusty yesterday an since then
I get wrong values for battery, fans, power consumption, power supply,
temperature, voltage (system is Dell PowerEdge R720xd):

----
sudo /usr/lib/nagios/plugins/check_ipmi_sensor -H localhost
IPMI Status: Critical [Presence = Critical, vFlash = Critical, Sensor #0
= Warning, N/A = N/A, Add-in Card 1 Presence = Critical, System Board
Intrusion = Critical, System Board Intrusion = Critical, System Board
Drive 0 = Critical, System Board Drive 0 = Critical, System Board
Intrusion = Critical, System Board Intrusion = Critical, System Board
Intrusion = Critical, System Board Intrusion = Critical, System Board
Intrusion = Critical, Power Supply 1 Status = Critical, Power Supply 1
Status = Warning] | 'Fan1'=3360.00;960.00:;720.00:
'Fan2'=3360.00;960.00:;720.00: 'Fan3'=3600.00;960.00:;720.00:
'Fan4'=4200.00;960.00:;720.00: 'Fan5'=4800.00;960.00:;720.00:
'Fan6'=4920.00;960.00:;720.00: 'Inlet Temp'=15.00;3.00:42.00;-7.00:47.00
'Exhaust Temp'=32.00;8.00:70.00;3.00:75.00 'Current 1'=0.40 'Current
2'=0.40 'Voltage 1'=228.00 'Voltage 2'=230.00 'Pwr
Consumption'=182.00;~:1792.00;~:1974.00 'Temp'=45.00 'Temp'=37.00
----

I've downgraded to 

  nagios-plugins-thomas-krenn_1.6-1~precise_all.deb

----
sudo /usr/lib/nagios/plugins/check_ipmi_sensor -H localhost
IPMI Status: Critical [Presence = Critical, vFlash = Critical] |
'Fan1'=3360.00;960.00:;720.00: 'Fan2'=3360.00;960.00:;720.00:
'Fan3'=3600.00;960.00:;720.00: 'Fan4'=4320.00;960.00:;720.00:
'Fan5'=4800.00;960.00:;720.00: 'Fan6'=4920.00;960.00:;720.00: 'Inlet
Temp'=15.00;3.00:42.00;-7.00:47.00 'Exhaust
Temp'=32.00;8.00:70.00;3.00:75.00 'Current 1'=0.40 'Current 2'=0.40
'Voltage 1'=228.00 'Voltage 2'=230.00 'Pwr
Consumption'=182.00;~:1792.00;~:1974.00 'Temp'=44.00 'Temp'=37.00
----

and it's working fine again.

Ciao
Marcus

tk-wfischer, 2016-05-27:
Hi Marcus,

reason is that now by default all SEL (system event log) entries are being checked - as soon as the SEL gets an entry, a warning/error is shown.

This is the according commit:
https://github.com/thomas-krenn/check_ipmi_sensor_v3/commit/588fdafecc6484f7c7046b4bb73e10b075af7310

Reason to change the behaviour was this issue:
https://github.com/thomas-krenn/check_ipmi_sensor_v3/issues/1

You can either use the --nosel option to disable SEL checks at all, or limit the type of sensors for which the SEL should be monitored with the -ST option, e.g: -ST 'Memory,Processor'

See also ./check_ipmi_sensors --help for details.

Can you check those options and give me feedback whether I can close this issue?

Best regards,
Werner

localguru, 2016-05-27:
Hi Werner,

Am Freitag, den 27.05.2016, 03:21 -0700 schrieb Werner Fischer:
> Hi Marcus,
> 
> reason is that now by default all SEL (system event log) entries are
> being checked - as soon as the SEL gets an entry, a warning/error is
> shown.
> 
> This is the according commit:
> 588fdaf
> 
> Reason to change the behaviour was this issue:
> #1
> 
> You can either use the --nosel option to disable SEL checks at all, or
> limit the type of sensors for which the SEL should be monitored with
> the -ST option, e.g: -ST 'Memory,Processor'
> 
> See also ./check_ipmi_sensors --help for details.
> 
> Can you check those options and give me feedback whether I can close
> this issue?
> 
Thanks for your really fast support.

Yes, with --nosel option it's working fine:

sudo /usr/lib/nagios/plugins/check_ipmi_sensor --nosel -H localhost -T
FAN
Sensor Type(s) FAN Status: OK | 'Fan1'=3360.00;960.00:;720.00:
'Fan2'=3360.00;960.00:;720.00: 'Fan3'=3600.00;960.00:;720.00:
'Fan4'=4200.00;960.00:;720.00: 'Fan5'=4800.00;960.00:;720.00:
'Fan6'=4920.00;960.00:;720.00:

I reinstalled version "nagios-plugins-thomas-krenn 1.7-2".

Thanks a lot
Marcus

tk-wfischer, 2016-05-27:
> Thanks for your really fast support.

You're welcome :-)
> Yes, with --nosel option it's working fine:

That's great news, thank you your feedback.

I'll now close this issue. In case you have any further question, just open a new issue again.

Best regards,
Werner

closed 2016-05-27 by tk-wfischer