thomas-krenn/check_ipmi_sensor_v3 #6: Provide -xT and -xST options to exclude sensor types from monitoring

tk-wfischer, 2016-06-07:
A customer sent me feedback that he wants to monitor simply all sensors, but no intrusion sensors. As the ID of the chassis intrusion sensor (e.g. on Supermicro based systems) may be different depending on the system, an option to exclude sensor types would be convenient. With such an option, a single check_ipmi_sensor configuration could be used to monitor all servers.

FreeIPMI provides options for ipmi-sensors and ipmi-sel (-T SENSOR-TYPE-LIST, --exclude-sensor-types=SENSOR-TYPE-LIST) for that.

I'll be looking into this topic and open hereby an issue to document the request.

gschoenberger, 2016-09-19:
Is -xx or -sx not sufficient?

> From the help text:
> Specify name and type pipe delimitered in this file to exclude an entry, for example:
> System Chassis Chassis Intru|Physical Security

There is no restriction to ID with -xx/-sx.

- Georg

tk-wfischer, 2016-09-19:
The customer (and also we ourselves) are looking for the 'exclude-sensor-types' option, so that the same options also works for boards from different vendors (where the name of the sensor might slightly differ for example).

The feature is still on our todo list, currently in the "next" column of our Kanban board :-) We keep you updated.

tk-wfischer, 2016-09-19:
https://github.com/tniedermeier and me just added the -xT option (see commit https://github.com/thomas-krenn/check_ipmi_sensor_v3/commit/69a283b03a515407efa9fdf631e703e23e07baa3). It worked well in our test:
`$ ./check_ipmi_sensor -H 10.1.102.130 -U xxxx -P xxxx -L xxxx --nosel -xT Physical_Security
IPMI Status: OK | 'System Temp'=24.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=26.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.66;0.54:1.49;0.51:1.52 '3.3VCC'=3.36;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.53;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-11.90;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.09;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.38;2.94:3.58;2.88:3.65
`

Without specifying -xT the IPMI status shows an error:
`$ ./check_ipmi_sensor -H 10.1.102.130 -U xxxx -P xxxx -L xxxx --nosel 
IPMI Status: Critical [Chassis Intru = Critical] | 'System Temp'=24.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=26.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.66;0.54:1.49;0.51:1.52 '3.3VCC'=3.36;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.52;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-11.90;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.09;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.38;2.94:3.58;2.88:3.65
`

We will keep working on -xST option :-)

veitw, 2016-09-19:
Hi,

how about an optional substitution parameter (e.g. a regexp) that
unifies the sensor names over all vendors' sensor names before actual
data processing takes place?

Example:

--renameSensor '^(ChassisIntru|Chassis_Intru|Physical_Security)$' 'Intrusion'

When reading the sensors list, this would rename all sensors either
named "ChassisIntru" or "Chassis_Intru" or "Physical_Security" to
"Intrusion" for further internal usage. Now for all systems the sensors
named "Intrusion" could be uniformly handled by any further steps such
as exclusion by name.

It also would enable people to to create uniform output and performance
data over a bunch of systems from different vendors or system models,
e.g. for combined per-rack intake air temperature graphs.

Regards,
// Veit

tk-wfischer, 2016-10-03:
Hi Veit, a renaming option to unify the output to create consistent performance data names is indeed a good idea. I have created a new issue for that: https://github.com/thomas-krenn/check_ipmi_sensor_v3/issues/8

tk-wfischer, 2016-10-03:
Just added the -xST option (see commit https://github.com/thomas-krenn/check_ipmi_sensor_v3/commit/d6075e2474f40a0f23fcec681d8bf9621b7d0f95). It worked well in our test:

`$ ./check_ipmi_sensor -H 10.1.102.130 -U xxxx -P xxxx -L xxxx -xST Physical_Security,OEM_Reserved,Session_Audit -xT Physical_Security
IPMI Status: OK | 'System Temp'=30.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=31.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.66;0.54:1.49;0.51:1.52 '3.3VCC'=3.36;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.52;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-12.09;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.09;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.38;2.94:3.58;2.88:3.65`

Without specifying -xST the IPMI status shows an error:
`$ ./check_ipmi_sensor -H 10.1.102.130 -U xxxx -P xxxx -L xxxx -xT Physical_Security
IPMI Status: Critical [Sensor #255 = N/A, Sensor #255 = N/A, Sensor #255 = N/A, Sensor #255 = Warning, Sensor #255 = Warning, System Chassis Chassis Intru = Critical] | 'System Temp'=30.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=31.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.68;0.54:1.49;0.51:1.52 '3.3VCC'=3.36;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.53;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-12.09;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.09;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.38;2.94:3.58;2.88:3.65`

In this example the SEL has the following entries:
`ID   | Date        | Time     | Name                                              | Type              | State    | Event`
`1    | Jan-01-1970 | 01:00:42 | Sensor #255                                       | OEM Reserved      | N/A      | Event Offset = 00h`
`2    | Jan-01-1970 | 01:00:41 | Sensor #255                                       | OEM Reserved      | N/A      | Event Offset = 00h`
`3    | Jan-01-1970 | 01:00:41 | Sensor #255                                       | OEM Reserved      | N/A      | Event Offset = 00h`
`4    | Jan-11-1970 | 23:37:43 | Sensor #255                                       | Session Audit     | Warning  | Invalid Username of Password`
`5    | Jan-11-1970 | 23:37:43 | Sensor #255                                       | Session Audit     | Warning  | Invalid Username of Password`
`6    | Sep-19-2016 | 15:09:13 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion`

PS: We'll continue on improving the SEL reporting function and will work on issue https://github.com/thomas-krenn/check_ipmi_sensor_v3/issues/7 now :-)

closed 2016-10-03 by tk-wfischer