thomas-krenn/check_ipmi_sensor_v3 #7: Provide more details on warnings causes by SEL entries

tk-wfischer, 2016-06-15:
Whenever there is an entry in the SEL (system event log), the plugin leads to a warning or error. Currently it is not always clear on the first view, why the plugin raises a warning.

This example shows one SEL entry, causing a warning (**Sensor #255 = Warning**) - the real reason which caused the warning is not shown:
```
$ ./check_ipmi_sensor -H 10.1.102.130 -U user -P relation -L user 
IPMI Status: Warning [Sensor #255 = Warning] | 'System Temp'=26.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=27.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.75;0.54:1.49;0.51:1.52 '3.3VCC'=3.34;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.53;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-11.90;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.07;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.36;2.94:3.58;2.88:3.65
```

The verbose output -v shows us more details (**Sensor #255 = Warning (Session Audit)**):
```
$ ./check_ipmi_sensor -H 10.1.102.130 -U user -P relation -L user -v
IPMI Status: Warning [Sensor #255 = Warning (Session Audit)] | 'System Temp'=26.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=28.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.72;0.54:1.49;0.51:1.52 '3.3VCC'=3.34;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.53;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-12.09;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.07;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.36;2.94:3.58;2.88:3.65
```

With very verbose -vvv we see the details (**Sensor #255, Session Audit, Warning, Invalid Username or Password**):
```
$ ./check_ipmi_sensor -H 10.1.102.130 -U user -P relation -L user -vvv
------------- debug output for sel (-vvv is set): ------------
  /usr/local/sbin/ipmi-sel was executed with the following parameters:
    /usr/local/sbin/ipmi-sel -h 10.1.102.130 -u user -p relation -l user --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all
  output of FreeIPMI:
ID   | Date        | Time     | Name                                              | Type              | State    | Event
1    | Jun-15-2016 | 10:24:45 | Sensor #255                                       | Session Audit     | Warning  | Invalid Username or Password
```

I suggest a default output like this whenever SEL entries causes warnings, showing the number of SEL entries and details only for the first or latest entry:
```
$ ./check_ipmi_sensor -H 10.1.102.130 -U user -P relation -L user 
IPMI Status: Warning [1 SEL entry present (Sensor #255 = Warning - Session Audit: Invalid Username or Password)] | 'System Temp'=26.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=27.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.75;0.54:1.49;0.51:1.52 '3.3VCC'=3.34;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.53;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-11.90;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.07;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.36;2.94:3.58;2.88:3.65
```

And an example with multiple SEL entries:
```
$ ./check_ipmi_sensor -H 10.1.102.130 -U user -P relation -L user 
IPMI Status: Warning [Chassis Intru = Critical ('General Chassis Intrusion'), 2 SEL entries present (Sensor #255 = Warning - Session Audit: Invalid Username or Password), for more entries see SEL or -vv verbose output] | 'System Temp'=26.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=27.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.75;0.54:1.49;0.51:1.52 '3.3VCC'=3.34;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.53;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-11.90;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.07;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.36;2.94:3.58;2.88:3.65
```

The verbose option -vv could include all SEL entries listed in multi-line output:
```
$ ./check_ipmi_sensor -H 10.1.102.130 -U user -P relation -L user -vv
IPMI Status: Warning [Chassis Intru = Critical ('General Chassis Intrusion'), 2 SEL entries present (Sensor #255 = Warning - Session Audit: Invalid Username or Password), for more entries see SEL or verbose output] | 'System Temp'=26.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=27.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.75;0.54:1.49;0.51:1.52 '3.3VCC'=3.34;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.53;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-11.90;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.07;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.36;2.94:3.58;2.88:3.65
System Temp = 26.00 (Status: Nominal)
Peripheral Temp = 28.00 (Status: Nominal)
CPU Temp = 'Low' (Status: Nominal)
FAN 1 = 2775.00 (Status: Nominal)
FAN 2 = 2700.00 (Status: Nominal)
FAN 3 = 2700.00 (Status: Nominal)
FAN 4 = 1050.00 (Status: Nominal)
Vcore = 0.66 (Status: Nominal)
3.3VCC = 3.34 (Status: Nominal)
12V = 12.14 (Status: Nominal)
VDIMM = 1.53 (Status: Nominal)
5VCC = 5.12 (Status: Nominal)
-12V = -12.09 (Status: Nominal)
VBAT = 3.07 (Status: Nominal)
VSB = 3.34 (Status: Nominal)
AVCC = 3.36 (Status: Nominal)
Chassis Intru = 'General Chassis Intrusion' (Status: Critical)
PS2 Status = 'Presence detected' (Status: Nominal)
SEL entries:
ID   | Date        | Time     | Name                                              | Type              | State    | Event
1    | Jun-15-2016 | 10:24:45 | Sensor #255                                       | Session Audit     | Warning  | Invalid Username or Password
2    | Jun-15-2016 | 10:50:05 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
```

ksuehring, 2016-06-16:
This issue is exactly what I was looking for when I just looked into the repository.

I recently started using the plugin and found it hard to distinguish between SEL entries and current sensor warnings. I'm also using the check_openmanage plugin for our Dell servers, which has a very similar output as the suggested one and makes the distinction very clear.

Unfortunately I have never worked with Perl and thus cannot help with the implementation.

tk-wfischer, 2016-10-03:
I have updated the code of the plugin. With commit https://github.com/thomas-krenn/check_ipmi_sensor_v3/commit/abc95595411549bc6fb3b5bab9e809bdacc3aedd the plugin shows a more comprehensible output for warnings caused by SEL entries.

The following example shows the normal output (withoug -v option):
`$ ./check_ipmi_sensor -H 10.1.102.130 -U xxxx -P xxxx -L xxxx
IPMI Status: Critical [9 system event log (SEL) entries present] | 'System Temp'=27.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=28.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.70;0.54:1.49;0.51:1.52 '3.3VCC'=3.36;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.53;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-11.90;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.09;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.38;2.94:3.58;2.88:3.65`

With -v the output looks like the following:
`$ ./check_ipmi_sensor -H 10.1.102.130 -U xxxx -P xxxx -L xxxx -v
IPMI Status: Critical [9 system event log (SEL) entries present - details: (Sensor #255 = N/A, OEM Reserved, Event Offset = 00h), (Sensor #255 = N/A, OEM Reserved, Event Offset = 00h), (Sensor #255 = N/A, OEM Reserved, Event Offset = 00h), (Sensor #255 = Warning, Session Audit, Invalid Username of Password), (Sensor #255 = Warning, Session Audit, Invalid Username of Password), (System Chassis Chassis Intru = Critical, Physical Security, General Chassis Intrusion), (System Chassis Chassis Intru = Critical, Physical Security, General Chassis Intrusion), (System Chassis Chassis Intru = Critical, Physical Security, General Chassis Intrusion), (System Chassis Chassis Intru = Critical, Physical Security, General Chassis Intrusion) - fix the reported issues and clear your SEL or exclude specific SEL entries using the -sx or -xST option] | 'System Temp'=27.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=28.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2775.00;600.00:18975.00;450.00:19050.00 'FAN 2'=2625.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=1050.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.66;0.54:1.49;0.51:1.52 '3.3VCC'=3.36;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.53;1.28:1.76;1.22:1.78 '5VCC'=5.12;4.58:5.34;4.32:5.60 '-12V'=-12.09;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.09;2.94:3.58;2.88:3.65 'VSB'=3.34;2.94:3.58;2.88:3.65 'AVCC'=3.38;2.94:3.58;2.88:3.65`

And here just for reference the -vvv part showing the output of ipmi-sel of my test machine:
`$ ./check_ipmi_sensor -H 10.1.102.130 -U xxxx -P xxxx -L xxxx -vvv
------------- debug output for sel (-vvv is set): ------------
  /usr/local/sbin/ipmi-sel was executed with the following parameters:
    /usr/local/sbin/ipmi-sel -h 10.1.102.130 -u xxxx -p xxxx -l xxxx --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all --exclude-sensor-types=
  output of FreeIPMI:
ID   | Date        | Time     | Name                                              | Type              | State    | Event
1    | Jan-01-1970 | 01:00:42 | Sensor #255                                       | OEM Reserved      | N/A      | Event Offset = 00h
2    | Jan-01-1970 | 01:00:41 | Sensor #255                                       | OEM Reserved      | N/A      | Event Offset = 00h
3    | Jan-01-1970 | 01:00:41 | Sensor #255                                       | OEM Reserved      | N/A      | Event Offset = 00h
4    | Jan-11-1970 | 23:37:43 | Sensor #255                                       | Session Audit     | Warning  | Invalid Username of Password
5    | Jan-11-1970 | 23:37:43 | Sensor #255                                       | Session Audit     | Warning  | Invalid Username of Password
6    | Sep-19-2016 | 15:09:13 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
7    | Oct-03-2016 | 17:34:47 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
8    | Oct-03-2016 | 18:23:11 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
9    | Oct-03-2016 | 18:24:36 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
[...]`

closed 2016-10-03 by tk-wfischer