thomas-krenn/check_ipmi_sensor_v3 #8: Option to rename sensors to create uniform performance data

tk-wfischer, 2016-10-03:
Provide an optional substitution parameter (e.g. a regexp) that unifies the sensor names over all vendors' sensor names before actual data processing takes place. This should provide consistent sensor names across systems from different vendors. With this feature, unified performance data could be created.

More details see Veit's posting here: https://github.com/thomas-krenn/check_ipmi_sensor_v3/issues/6#issuecomment-247977629

widhalmt, 2018-12-18:
Hi,

Is there any progress on this? I tried to use this plugin for some other hardware vendor who just changed the names of their sensors with a newer generation of their hardware.

Cheers,
Thomas

gschoenberger, 2019-01-11:
Commit 00b0c3207a73e90743bf15a928c4ccde09070cb8 adds this feature, text from --help:
use an unify file to unify sensor names. This is an easy way to rename
```
  [-us|--unify-sensors <unify file>]
       use an unify file to unify sensor names. This is an easy way to rename
       sensors with given patterns in the file. Once might us this option
       to get the same sensor names accross different platforms, e.g. to only
       have 'intrusion' as sensor name and not 'ChassisIntru', 'Chassis_Intru'
       or 'Chassis Intru'. Rules in the file follow simple regex patterns e.g.:
       ^(Chassis Intru|ChassisIntru|Chassis_Intru|Physical_Security)$/Intrusion
       Temp$/TEMP
```

gschoenberger, 2019-01-11:
Happy Testing!

tniedermeier, 2019-01-11:
Works great!
./check_ipmi_sensor -H ipmi-address -U adminuser -P admin-password -L admin -us unify-file

IPMI Status: Warning [3 system event log (SEL) entries present] | 'CPU TEMP'=28.00;0.00:95.00;0.00:100.00 'PCH TEMP'=28.00;16.00:90.00;5.00:95.00 'Mainboard Temperature'=23.00;0.00:80.00;-5.00:85.00 'Peripheral TEMP'=32.00;0.00:80.00;-5.00:85.00 'VcpuVRM TEMP'=29.00;5.00:95.00;0.00:100.00 'DIMMA1 TEMP'=25.00;5.00:80.00;0.00:85.00 'FAN1'=5600.00;700.00:25300.00;500.00:25400.00 'FAN2'=5600.00;700.00:25300.00;500.00:25400.00 '12V'=12.19;10.78:12.96;10.27:13.28 '5VCC'=4.97;4.48:5.39;4.30:5.55 '3.3VCC'=3.37;2.96:3.55;2.82:3.66 'VBAT'=2.94;2.58:3.48;2.50:3.59 'VCPU'=0.11;0.08:1.52;0.08:1.52 'VDIMMAB'=1.21;1.05:1.34;0.97:1.43 '5VSB'=4.92;4.48:5.39;4.38:5.55 '3.3VSB'=3.30;2.96:3.55;2.89:3.66 'VBMC 1.2V'=1.20;1.09:1.34;1.05:1.37 'VPCH 1.0V'=0.99;0.90:1.14;0.86:1.16

unity-file:

^(Chassis Intru|ChassisIntru|Chassis_Intru|Physical_Security)$/Intrusion
^(MB1 Temperature|System Temp)$/Mainboard Temperature
Temp$/TEMP

gschoenberger, 2019-01-11:
Great!

closed 2019-01-11 by gschoenberger