thomas-krenn/check_ipmi_sensor_v3 #10: Missing backplane?

pirx242, 2017-01-16:
Hello!

Not an "issue" but i dont know where to look for usage help:)

Running the script on a R630 Dell server, and i get this (see below)

How can there be two backplanes?
How can i ignore this somehow?
When i log in to the DRAC of the server over https, i can see no problem.

Any idea? 

  wbr / Alex

./check_ipmi_sensor -H server-drac -U <root> -P <pwd> -L Administrator -vv
IPMI Status: Critical [Presence = Critical ('Entity Absent'), BP1 Presence = Critical ('Entity Absent')] | 'Fan1A RPM'=3240.00 'Fan2A RPM'=3240.00 'Fan3A RPM'=3240.00 'Fan4A RPM'=3240.00 'Fan5A RPM'=3360.00 'Fan6A RPM'=3360.00 'Inlet Temp'=17.00 'CPU Usage'=0.00 'IO Usage'=0.00 'MEM Usage'=0.00 'SYS Usage'=1.00 'Exhaust Temp'=31.00 'Temp'=42.00 'Temp'=39.00 'Fan7A RPM'=3240.00 'Fan1B RPM'=3120.00 'Fan2B RPM'=3000.00 'Fan3B RPM'=3000.00 'Fan4B RPM'=3000.00 'Fan5B RPM'=3000.00 'Fan6B RPM'=3000.00 'Fan7B RPM'=3120.00 'Current 1'=0.40 'Current 2'=0.20 'Voltage 1'=234.00 'Voltage 2'=232.00 'Pwr Consumption'=84.00
Intrusion = 'OK' (Status: Nominal)
Fan1A RPM = 3240.00 (Status: Nominal)
Fan2A RPM = 3240.00 (Status: Nominal)
Fan3A RPM = 3240.00 (Status: Nominal)
...
BP1 5V PG = 'State Deasserted' (Status: Nominal)               <---- ?
BP2 5V PG = 'State Deasserted' (Status: Nominal)
...
BP1 Presence = 'Entity Absent' (Status: Critical)               <---- ?
BP2 Presence = 'Entity Present' (Status: Nominal)
...

tk-wfischer, 2017-01-16:
Hi Alex,

thank you for using the check_ipmi_sensor plugin and for asking your question.

It could be that this BMC firmware is also used for other servers/chassis which contain two backplanes, and that this fact causes your issue. We have implemented the -x option for such cases to ignore sensors like that.

You can find more information on that also in the help text (using --help):
```
  [-x <sensor id>]
       exclude sensor matching <sensor id>. Useful for cases when unused
       sensors cannot be deleted from SDR and are reported in a non-OK state.
       Option can be specified multiple times. The <sensor id> is a numeric
       value (sensor names are not used as some servers have multiple sensors
       with the same name). Use -vvv option to query the <sensor ids>.
```

Please try using the -x option and let us know whether this solves your issue or whether you have some further questions (we are happy to help).

I hope this information is helpful for you.
Good look,
Werner

pirx242, 2017-01-16:
Ah, awsome, thanks for fast answer! :)

I have already tried the -x option. But i didnt know what argument i should specify...
Tried -x BP1 and -x "BP1 Presence", but none of those two worked.

Do you have any other suggestion? Or how to get some list of sensors?

tk-wfischer, 2017-01-16:
You're welcome :-)
As mentioned in the help text, you can use the -vvv (3 times the "v") option to get the numerical sensor id.

So in your case this would be:
```
./check_ipmi_sensor -H server-drac -U -P -L Administrator -vvv
```

pirx242, 2017-01-16:
Aah, those IDs scrolled off screen before i noticed them, but i found them now!
Thanks a lot!

closed 2017-01-16 by pirx242