thomas-krenn/check_ipmi_sensor_v3 #12: Provide only system event log

sma-inberlin, 2017-03-19:
Hi Thomas,

it would be nice if we can have a option to specify that only system event log are printed. I'd like to separate Icinga services. 

Thanks

marcelfischer, 2017-09-08:
Any news on this? Would be great to include or exclude event log

tniedermeier, 2017-09-08:
Hi,
thanks for your message. It's still on our agenda, but you can use a tiny tweak to achieve only the output of the SEL. I first executed check_ipmi_sensors with maximum verbosity (-vvv) to get all sensor types (column 3 "Type" in the table with all the sensors) and afterwards I excluded, using the -xT Parameter, these sensor types. So the result was only one line of output with just the SEL information.
--> /usr/lib/nagios/plugins/check_ipmi_sensor -H <IP-Address> -U <Username> -P <Password> -L user -xT Temperature,Voltage,Fan

Maybe you could give this workaround a try, in my tests it was sufficient to exclude this three types of sensors.

Best regards,
Thomas

marcelfischer, 2017-09-08:
Hi Thomas,

yes, this could be workaround. But I have much more types than those three. And the problem is that, if you exclude a sensor type, that does not exists for a specific server, you get an error like "invalid sensor type". So I need to configure many exceptions...

But for now at least I added --nothresholds to avoid get this many values I dont need.

d4nys3k, 2019-01-04:
Hello,
i tried to create patch implementing this function - checking only SEL (--selonly). There's also another option (--seltail) allowing limit number of log messages obtained by ipmi-sel command. 

[check_ipmi_sensor.patch.txt](https://github.com/thomas-krenn/check_ipmi_sensor_v3/files/2727534/check_ipmi_sensor.patch.txt)

Hope it helps / will be merged sometimes :)

gschoenberger, 2019-01-04:
Just to ensure the patch is correct, can you also attach the current complete version of your plugin file?

THX - Georg

d4nys3k, 2019-01-04:
> Just to ensure the patch is correct, can you also attach the current complete version of your plugin file?
Here's complete script: [check_ipmi_sensor.txt](https://github.com/thomas-krenn/check_ipmi_sensor_v3/files/2728110/check_ipmi_sensor.txt)

tniedermeier, 2019-01-08:
Thank you @d4nys3k for your files, I have tested it with a Supermicro X11SSH-LN4F, looks good!

./check_ipmi_sensor -H <IPMI-IP> -U <USER> -P <PASSWORD> -L admin --selonly
SEL Status: Warning [3 system event log (SEL) entries present]

./check_ipmi_sensor -H <IPMI-IP> -U <USER> -P <PASSWORD> -L admin --seltail 1
IPMI Status: Warning [1 system event log (SEL) entry present] | 'CPU Temp'=41.00;0.00:95.00;0.00:100.00 'PCH Temp'=31.00;16.00:90.00;5.00:95.00 'System Temp'=23.00;0.00:80.00;-5.00:85.00 'Peripheral Temp'=34.00;0.00:80.00;-5.00:85.00 'VcpuVRM Temp'=36.00;5.00:95.00;0.00:100.00 'DIMMA1 Temp'=25.00;5.00:80.00;0.00:85.00 'FAN1'=5600.00;700.00:25300.00;500.00:25400.00 'FAN2'=5600.00;700.00:25300.00;500.00:25400.00 '12V'=12.13;10.78:12.96;10.27:13.28 '5VCC'=4.95;4.48:5.39;4.30:5.55 '3.3VCC'=3.37;2.96:3.55;2.82:3.66 'VBAT'=2.92;2.58:3.48;2.50:3.59 'VCPU'=1.03;0.08:1.52;0.08:1.52 'VDIMMAB'=1.21;1.05:1.34;0.97:1.43 '5VSB'=4.92;4.48:5.39;4.38:5.55 '3.3VSB'=3.30;2.96:3.55;2.89:3.66 'VBMC 1.2V'=1.20;1.09:1.34;1.05:1.37 'VPCH 1.0V'=0.99;0.90:1.14;0.86:1.16

Best regards
Thomas

tniedermeier, 2019-01-08:
Verbose output also looks fine (-vvv) with both parameters.

gschoenberger, 2019-01-10:
THX @d4nys3k - the patch is rebased and live in the master branch!

All the best, Georg

closed 2019-01-10 by gschoenberger