thomas-krenn/check_ipmi_sensor_v3 #13: clarification on the -fc option

sandrotosi, 2017-05-30:
Hello,
reading the help page i can see:

```  [-fc <num fans>]
       number of fans that should be active. If the number of current active
       fans reported by IPMI is smaller than <num fans> then a Warning state
       is returned.
```

which i understand it as the number of *working* fans (ie "active"), but reading the code:

```
                        if( $row->{'type'} eq 'Fan' && $row->{'reading'} ne 'N/A' ){
                                $curr_fans++;
                        }
...
                #now check if num fans equals desired unit fans
                if( $fan_count ){
                        if( $curr_fans < $fan_count ){
                                $exit = 1 if $exit < 1;
```

which means -fc counts the number of *installed* fans.

can you clarify what is the actual logic of this option: installed of working fans?

thanks!

veitw, 2017-05-30:
Am Dienstag, den 30.05.2017, 09:30 -0700 schrieb Sandro Tosi:
> can you clarify what is the actual logic of this option: installed of
> working fans?

Hi Sandro,

this is not my code, but I am quite sure that this especially handles
"fan died while offline" situations:

When a fan dies or is disconnected while the IPMI controller ist powered
on, the fan will be reported as failed and running at 0 rpm.

But when the IPMI controller is rebooted and/or an inventory survey is
executed with a failed fan, or a fan is disconnected while the system is
completely powered off (e.g. pulling a connector while servicing a
machine), most IPMI BMCs will report the fan as not installed instead of
failed.

Therefore, IMO, the code handles this situation well. But it might be
useful to update the documentation from "working" to "installed".

Best regards,
// Veit

gschoenberger, 2017-06-02:
Hi Veit,
thanks for clarification, I totally agree with you. The option is simply for "hey my server has 4 fans installed, and I want to be notified if IPMI reports a different number than 4".
We should definitively update the documentation,
- Georg

sandrotosi, 2017-06-05:
i dont think it works exactly as you expect it to: if you have N installed fans but M (with M > N) fan sensors, than even using -fc will result and IPMI error due to the missing readings form the M-N fan sensors

gschoenberger, 2019-01-10:
I have updated the help text with 47a0b5efebb7938facd5bc06205c7dfd9b3f7e4e

closed 2019-01-10 by gschoenberger