thomas-krenn/check_ipmi_sensor_v3 #16: Plugin throws WARN when first FAN does not start with "1"

ronator, 2017-06-27:
Hi,

I monitor some HP ILO interfaces with this plugin and help of freeipmi in icinga2. It works great!

Today, I tried to implement this check for a different HP server and got a FAN Warning:

IPMI Status: Warning [Fan = Warning] 

I found out, that this system has only 4 fans configured: 'Fan 3' up to 'Fan 6'. I tried to set '-fc 4' but that causes still a warning. I _guess_, the plugin expects fan numbers to start with '1'.

We do not plan to add 'Fan 1' and 'Fan 2' and the system is running just fine. So my question is:

Is or will there be a way to set something like a 'fan offset" so that the plugin will not warn if fan numbers start with something else than '1'? I am assuming that's causing the warning, but not 100% sure.

/usr/lib/nagios/plugins/check_ipmi_sensor -H 192.168.xxx.xxx -f /etc/freeipmi/icinga.conf -fc 4

IPMI Status: Warning [Fan = Warning] 

**Thanks in advance.**

tk-wfischer, 2017-07-07:
Hi ronator,

the -fc option only adds another option to verify the number of active fans: _number of fans that should be active. If the number of current active fans reported by IPMI is smaller than <num fans> then a Warning state is returned._

Please post the verbose output (-vvv option), than I can give you further hints how to solve your issue. I think you will need to add the -x option to exclude sensors for Fan 1 and/or Fan 2. But with the verbose output I can give you a concrete detailed information :-)

ronator, 2017-07-11:
Hi and yes, let me check that right away.

ronator, 2017-07-11:
I did not know you can exclude some fans with -x. 

/usr/lib/nagios/plugins/check_ipmi_sensor -H 192.168.x.x -f /etc/freeipmi/icinga.conf -fc 4 -vvv

The verbose output is the following:
```
------------- debug output for sel (-vvv is set): ------------
  /usr/sbin/ipmi-sel was executed with the following parameters:
    /usr/sbin/ipmi-sel -h 192.168.x.x --config-file /etc/freeipmi/icinga.conf --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all --exclude-sensor-types=
  output of FreeIPMI:
ID | Date        | Time     | Name                                             | Type         | State    | Event
5  | Jan-26-2015 | 10:31:28 | Sensor #0                                        | OS Critical Stop | Warning  | OS Graceful Shutdown
6  | Jan-26-2015 | 10:31:28 | N/A                                              | N/A              | N/A      | OEM defined = 00h 03h 00h 02h 80h 00h
7  | Jan-26-2015 | 10:34:00 | Sensor #0                                        | OS Boot          | Nominal  | C: boot completed
8  | Jan-26-2015 | 10:34:00 | N/A                                              | N/A              | N/A      | OEM defined = 00h FFh 09h C6h 54h 00h
9  | Jan-26-2015 | 10:34:41 | Sensor #0                                        | OS Critical Stop | Warning  | OS Graceful Shutdown
10 | Jan-26-2015 | 10:34:41 | N/A                                              | N/A              | N/A      | OEM defined = 00h 03h 00h 02h 80h 00h
11 | Jan-26-2015 | 10:36:55 | Sensor #0                                        | OS Boot          | Nominal  | C: boot completed
12 | Jan-26-2015 | 10:36:55 | N/A                                              | N/A              | N/A      | OEM defined = 00h B5h 0Ah C6h 54h 00h
13 | Jan-26-2015 | 10:40:59 | Sensor #0                                        | OS Critical Stop | Warning  | OS Graceful Shutdown
14 | Jan-26-2015 | 10:40:59 | N/A                                              | N/A              | N/A      | OEM defined = 00h FFh 00h 05h 00h 00h
15 | Jan-26-2015 | 10:43:17 | Sensor #0                                        | OS Boot          | Nominal  | C: boot completed
16 | Jan-26-2015 | 10:43:17 | N/A                                              | N/A              | N/A      | OEM defined = 00h 33h 0Ch C6h 54h 00h
17 | Jan-26-2015 | 10:45:48 | Sensor #0                                        | OS Critical Stop | Warning  | OS Graceful Shutdown
18 | Jan-26-2015 | 10:45:48 | N/A                                              | N/A              | N/A      | OEM defined = 00h 02h 00h 04h 84h 00h
19 | Jan-26-2015 | 10:48:09 | Sensor #0                                        | OS Boot          | Nominal  | C: boot completed
20 | Jan-26-2015 | 10:48:09 | N/A                                              | N/A              | N/A      | OEM defined = 00h 55h 0Dh C6h 54h 00h
21 | Jan-26-2015 | 10:48:35 | Sensor #0                                        | OS Critical Stop | Warning  | OS Graceful Shutdown
22 | Jan-26-2015 | 10:48:35 | N/A                                              | N/A              | N/A      | OEM defined = 00h 03h 00h 02h 80h 00h
23 | Jan-26-2015 | 10:50:56 | Sensor #0                                        | OS Boot          | Nominal  | C: boot completed
24 | Jan-26-2015 | 10:50:56 | N/A                                              | N/A              | N/A      | OEM defined = 00h FDh 0Dh C6h 54h 00h
25 | Jan-26-2015 | 11:49:44 | Sensor #0                                        | OS Critical Stop | Warning  | OS Graceful Shutdown
26 | Jan-26-2015 | 11:49:44 | N/A                                              | N/A              | N/A      | OEM defined = 00h 01h 00h 01h 80h 00h
27 | Jan-26-2015 | 11:49:44 | N/A                                              | N/A              | N/A      | OEM defined = 01h 52h 00h 65h 00h 00h
28 | Jan-26-2015 | 11:49:45 | N/A                                              | N/A              | N/A      | OEM defined = 02h 62h 00h 6Fh 00h 00h
29 | Jan-26-2015 | 11:49:45 | N/A                                              | N/A              | N/A      | OEM defined = 03h 6Fh 00h 74h 00h 00h
30 | Jan-26-2015 | 11:49:45 | N/A                                              | N/A              | N/A      | OEM defined = 04h 69h 00h 6Eh 00h 00h
31 | Jan-26-2015 | 11:49:45 | N/A                                              | N/A              | N/A      | OEM defined = 05h 67h 00h 20h 00h 00h
32 | Jan-26-2015 | 11:49:45 | N/A                                              | N/A              | N/A      | OEM defined = 06h 61h 00h 74h 00h 00h
33 | Jan-26-2015 | 11:49:45 | N/A                                              | N/A              | N/A      | OEM defined = 07h 20h 00h 48h 00h 00h
34 | Jan-26-2015 | 11:49:45 | N/A                                              | N/A              | N/A      | OEM defined = 08h 50h 00h 20h 00h 00h
35 | Jan-26-2015 | 11:49:45 | N/A                                              | N/A              | N/A      | OEM defined = 09h 53h 00h 55h 00h 00h
36 | Jan-26-2015 | 11:49:45 | N/A                                              | N/A              | N/A      | OEM defined = 0Ah 4Dh 00h 20h 00h 00h
37 | Jan-26-2015 | 11:49:45 | N/A                                              | N/A              | N/A      | OEM defined = 0Bh 75h 00h 73h 00h 00h
[... cut ...]
C1 P1I Bay 1 = 'Drive Presence' (Status: Nominal)
C1 P1I Bay 2 = 'Drive Presence' (Status: Nominal)
C1 P1I Bay 3 = 'Drive Presence' (Status: Nominal)
C1 P1I Bay 4 = 'Drive Presence' (Status: Nominal)
C1 P2I Bay 5 = 'Drive Presence' (Status: Nominal)
C1 P2I Bay 6 = 'Drive Presence' (Status: Nominal)
```

As you can see the fan numbering starts with "Fan 3" and we have 4 summed up. Please explain to me how I can exclude the first two non-present fans. I assume I have to get the sensor IDs for
```
[-x <sensor id>]
```
Thanks a lot for your help.

ronator, 2017-07-25:
I forgot to answer, I had this site opened as a tab for like two weeks and just found it - my bad.

tk-wfischer, 2017-07-25:
Hi,
sorry also from my side for overlooking your answer 14days ago.

From your current debug output, it seems that only the SEL (system event log) entries are causing the critical state ("58 system event log (SEL) entries present").

You have two options:

1. Do not monitor SEL entries by using the "--nosel" option of the plugin
2. Delete the SEL entries (e.g. by FreeIPMI's "ipmi-sel --clear" - see https://www.gnu.org/software/freeipmi/manpages/man8/ipmi-sel.8.html or by "ipmitool sel clear" or within the Web Interface of your BMC.

I would recommend option 2. The reason is that there could be possible errors with many servers do not monitor with the normal sensors reported by "ipmi-sensors", but which can cause new entries in the system event log. An example for this kind of errors are e.g. Memory warnings like "DIMM3 correctable ECC error detected" or the like.

gschoenberger, 2018-11-29:
Closing this due to inactivity

closed 2018-11-29 by gschoenberger