thomas-krenn/check_ipmi_sensor_v3 #17: Relaxed matching of Removed/Absent Device/Entity

flohoff, 2017-08-11:
Hi,
please include this patch - it relaxes the matching for absent/removed devices/entitys a bit.

diff --git a/check_ipmi_sensor b/check_ipmi_sensor
index a735b7e..fcc95fd 100755
--- a/check_ipmi_sensor
```
+++ b/check_ipmi_sensor
@@ -764,7 +764,7 @@ MAIN: {
                                $row->{'name'} =~ s/ /_/g;
                        }
                        my $check_sensor_state = 1;
-                       if($no_entity_absent && ($row->{'event'} eq '\'Entity Absent\'')){
+                       if($no_entity_absent && ($row->{'event'} =~ /\'.*((Device|Entity) (Absent|Removed)).*\'/)) {
                                $check_sensor_state = 0;
                        }
                        #check for warning sensors

```

gschoenberger, 2018-03-02:
Can you test commit 68588829106159984c9b528052979725145723f0 ?

flohoff, 2018-03-13:
I left the company and dont have access to Supermicro/TK Servers anymore - but i notified my former collegues to check and update their internal tools.

gschoenberger, 2018-10-16:
Closing due to inactivity and the commit is already merged...

closed 2018-10-16 by gschoenberger

taupehat, 2020-06-24:
I'm not seeing this code any longer, and have Supermicro systems which are giving me exactly this sort of grief. Am willing to help...