thomas-krenn/check_ipmi_sensor_v3 #18: noentityabsent does not work in conjunction with ipmimonitoring legacy-output

jaykumar-jnpr, 2018-02-28:
/usr/home/_zenoss/check_ipmi_sensor.pl --nosel -o zenoss --nothresholds --noentityabsent -vvv
------------- debug output for sensors (-vvv is set): ------------
  script was executed with the following parameters:
    /usr/home/_zenoss/check_ipmi_sensor.pl --nosel -o zenoss --nothresholds --noentityabsent -vvv
  check_ipmi_sensor version:
    3.12
  FreeIPMI version:
    ipmimonitoring - 0.8.12
  FreeIPMI was executed with the following parameters:
    /usr/sbin/ipmimonitoring --quiet-cache --sdr-cache-recreate --interpret-oem-data --legacy-output
  FreeIPMI return code: 0
  output of FreeIPMI:

88 | DASD Backplane 1 | Entity Presence | Nominal | N/A | 'Entity Present'
89 | DASD Backplane 2 | Entity Presence | Nominal | N/A | 'Entity Present'
90 | DASD Backplane 3 | Entity Presence | Critical | N/A | 'Entity Absent'
91 | DASD Backplane 4 | Entity Presence | Critical | N/A | 'Entity Absent'

Error out with

Use of uninitialized value in string eq at /usr/home/_zenoss/check_ipmi_sensor.pl line 761.

gschoenberger, 2018-03-02:
Could you please post the full output of:
`/usr/sbin/ipmimonitoring --quiet-cache --sdr-cache-recreate --interpret-oem-data --legacy-output`

THX Georg

jaykumar-jnpr, 2018-03-02:
Here is the output [ipmimonitoring](https://gist.github.com/jaykumar-jnpr/0e9545ffbd40e769aea311107834ea19)

gschoenberger, 2018-03-02:
OK, I think I found the problem.
Your output is missing the "Event" header row which is used to check the entity presence:
```
Record ID | Sensor Name | Sensor Group | Monitoring Status | Sensor Units | Sensor Reading
1 | SysBrd 3.3V | Voltage | Nominal | V | 3.308600
2 | SysBrd 5V | Voltage | Nominal | V | 5.085000
[...]
88 | DASD Backplane 1 | Entity Presence | Nominal | N/A | 'Entity Present'
89 | DASD Backplane 2 | Entity Presence | Nominal | N/A | 'Entity Present'
90 | DASD Backplane 3 | Entity Presence | Critical | N/A | 'Entity Absent'
91 | DASD Backplane 4 | Entity Presence | Critical | N/A | 'Entity Absent'
```
Lines 767 - 769 does the check, but you don't have an event column:
```
if($no_entity_absent && ($row->{'event'} eq '\'Entity Absent\'')){
	$check_sensor_state = 0;
}
```
In your case I think we should use "Sensor Reading" which is mapped to `$row->{'reading'}`.
Maybe we should map "Sensor Reading" to `$row->{'event'}` if the Event header row is missing, but I am not sure if we can do that in general.
I think I will discuss that with @tk-wfischer .

Does the rest of the plugin work or do you have other problems too?

jaykumar-jnpr, 2018-03-02:
I am implementing this plugin in an heterogenous environment with multiple Vendor/Platform. It seems to working fine, except for above issue specifically on IBM Chassis.

I will open issues for any other bug I find.

jaykumar-jnpr, 2018-11-28:
@tk-wfischer @gschoenberger 

Would appreciate if you could take a look into this issue.

gschoenberger, 2018-11-29:
Can you test commit b4ea3bae3ab9bf9e1cb50319111975f8fe532f11?

jaykumar-jnpr, 2018-11-29:
Worked like a charm...

For record - [ipmimonitoring](https://gist.github.com/jaykumar-jnpr/f3e338dc275a60543286a4076ecb59fc)

Please close the Issue

closed 2018-11-30 by gschoenberger