thomas-krenn/check_ipmi_sensor_v3 #22: Warning notification from Cisco USC Systems

omahonyb, 2018-07-16:
I am running check_ipmi_sensor [v3] with "--noentityabsent" on some Cisco hardware. However I get this for PSU1&2:
PSU1_STATUS = 'Presence detected' (Status: Nominal)
PSU1_AC_OK = 'State Asserted' (Status: Warning)
PSU1_PWRGD = 'State Asserted' (Status: Warning)
PSU1_VOUT = 11.90 (Status: Nominal)
PSU1_IOUT = 5.00 (Status: Nominal)
PSU1_POUT = 64.00 (Status: Nominal)

My SEL is clear, and in the ICM [Cisco Integrated Management Controller] everything is green. Is there any way to resolve this other than exclude with -x ?

tk-wfischer, 2018-07-16:
It seems to me that your Cisco system uses here an IPMI sensor type in a "unusual" way. The standard types would be:
https://github.com/chu11/freeipmi-mirror/blob/master/etc/freeipmi_interpret_sensor.conf#L267

Al Chu mentions there:
```
# Most default interpretations can be determined quite easily and can
# meet the needs of most users.  For example, a reading of
# "Performance_Met" is normally better than "Performance_Lags".
# However, some sensors can be ambiguous and depend completely on the
# manufacturer.  For example, "State_Asserted" vs. "State_Deasserted"
# are completely at the interpretation of the vendor. Users are
# advised to adjust the interpretations below as needed.
```
So I'd suggest that you run ipmi-sensors with the verbose option -vv to get some more details about those two sensors - see https://www.thomas-krenn.com/de/wiki/FreeIPMI_ipmi-sensors#ipmi-sensors_-r_ID_-vv for an example. You can share the output and I'll be happy to help.

Than you could change the behaviour how FreeIPMI should interpret those two sensors PSU1_AC_OK and PSU1_PWRGD. This would be the best solution, as it seems to me that those two sensors should be in the state "asserted" when everything is fine and that you should receive a warning once one of those two sensors is de_asserted.

omahonyb, 2018-07-16:
Why am I not surprised Cisco is doing things in a stupid way :) [This is happening on 3 out of 5 different HW types]

I know very little about ipmi, so any help would be gratefully accepted, and also thanks for the prompt response!
`root@XXXX15:27:56 ~ # ipmi-sensors -r 40 -vv
Record ID: 40
Record Type: Full Sensor Record (1h)
ID String: PSU1_AC_OK
Sensor Type: Power Supply (8h)
Sensor Number: 39
IPMB Slave Address: 10h
Sensor Owner ID: 20h
Sensor Owner LUN: 0h
Channel Number: 0h
Entity ID: power supply (10)
Entity Instance: 1
Entity Instance Type: Physical Entity
Event/Reading Type Code: 3h
Accuracy: 0.000000%
Sensor Direction: Unspecified
Assertion Event Enabled: 'State Deasserted'
Assertion Event Enabled: 'State Asserted'
Deassertion Event Enabled: 'NONE'
Sensor Event: 'State Asserted'`

gschoenberger, 2018-07-16:
It seems the warning is generated from "Sensor Event: 'State Asserted'", but as Al Chu states these event states are completely vendor specific. I don't think we can handle these states in the plugin, @tk-wfischer what do you think?

omahonyb, 2018-07-17:
@gschoenberger 

If that is the case, no worries. I'll figure out a method of ignoring them based on machine type. From my reading of the link TK put in, it seems Cisco are doing things arseways with those checks. Asserted to them is fine, but it is being flagged as Warning by freeipmi. Is there a way to write a specific conf file for these [or should I just go over there and ask?]

tk-wfischer, 2018-07-20:
@gschoenberger: you are right - we cannot (and do not want to) handle this by the plugin

@omahonyb: yes, it seems that Cisco is using this IPMI sensor in a wrong way. According to https://www.intel.com/content/dam/www/public/us/en/documents/product-briefs/ipmi-second-gen-interface-spec-v2-rev1-1.pdf chapter "42.2 Sensor Type Codes and Data", page 506 (page 532 of the PDF) the offset "03h" - which is your "Event/Reading Type Code: 3h" is used for "Power Supply input lost (AC/DC)".
I'm not aware of a possibility to change the behaviour of FreeIPMI here. I'd suggest to ask Cisco about that.

tk-wfischer, 2018-07-20:
Just one addition: you could ask your question on the FreeIPMI mailing list: https://lists.gnu.org/mailman/listinfo/freeipmi-users

Maybe there is a option I'm not aware of or I may have overlooked something.

closed 2018-07-23 by gschoenberger