thomas-krenn/check_ipmi_sensor_v3 #24: problem creating rpm under redhat 6

tlo8640, 2018-10-12:
Hello,

I tried to create rpms from this project for rhel 6 and 7.
It worked for rhel7, but it failed for rhel 6, because of a strange problem of automatic dependency resolution for perl modules in the rpm build process.
This feature (automatic dependency resolution) tries to create a dependency list of perl modules for the package for installing them when you will install the built rpm package on a system. I don't know how exactly this feature works, but somehow it scans the code for the "use" keyword and takes the word after use as a needed module name and add this to the dependency list. 
If you grep your script for "use" at the beginning of a line you will find some lines:
` egrep '^[[:blank:]]*use' check_ipmi_sensor
use strict;
use warnings;
use Getopt::Long qw(:config no_ignore_case);
use IPC::Run qw( run ); #interact with processes
         username monitoring
       use a sel exclude file to exclude entries from the system event log.
       use an exclude file to exclude sensors.
use of this software. The mailing list is available at
`

the created rpm cannot be installed on rhel 6:

`Error: Package: check_ipmi_sensor-3.12-3.el6.noarch 
           Requires: perl(a)
Error: Package: check_ipmi_sensor-3.12-3.el6.noarch 
           Requires: perl(an)`

Perhaps it's a problem of the rpm macros for redhat 6 only, because it works with redhat 7. Disabling automatic dependency resolution solves this proble. But If you want this, you have to move the "use" away from the beginning of the line or replace it with another word (e.g. "take"). But only where it is not the part of the perl code! :-)

Regards
Thomas

P.S. Thank you for this script, it helps us a lot!

tlo8640, 2018-10-15:
the following line in spec file fixed the wrong requirements for me:
````bash
%if 0%{?el6}
%filter_from_requires /perl(a[n]*)/d
%filter_setup
%endif
````

tlo8640, 2018-10-15:
Hello,

yes, at least it worked fine for rhel6, too.
So, it's not a bug in your code, more an info for other people who need/want to built rpms from your code.
Thanks and regards
Thomas

gschoenberger, 2018-10-16:
Can you post the full spec file? Maybe we should add it to the master branch as additional info,
THX

tniedermeier, 2019-01-08:
Thank you @tlo8640 also from my side for providing the spec file, I added it with commit a00ce9ee0e5813879335b807b21ae6c81de581c7.

Best regards
Thomas

closed 2019-01-08 by tniedermeier