thomas-krenn/check_ipmi_sensor_v3 #25: Wildcard in Sel Exclude file

omahonyb, 2018-11-23:
Hey mate. These annoying Cisco UCS servers are causing me grief again. Due to their sel logging on boot for various items superfluously logged as issues, we have begun using the -sx option for a sel file. It is all working great thanks, except one issue.
Cisco log their ECC error checking of DIMM modules to the sel as expected. However, these are logged as:
Memory Module 0 DDR4_P1_A1_ECC| Memory
With the '0' value and 'P1_A1' values changing between server hardware versions.
Is there any way to specify all slots?

gschoenberger, 2018-11-29:
Currently it is only possible to exclude a specific SEL sensor type:
```
  [-xST <SEL sensor type(s)]
       exclude SEL entries of specific sensor types.
       Multiple sensor types can be specified as a comma-separated list.
```
Would this be an option for you? E.g. exclude the Memory sensors from the SEL in general...

omahonyb, 2018-11-30:
I am waiting for an actual DIMM error, but I am afraid that if we ignore the Memory correctable errors, we might miss real HW errors.
Would have been nice to have a threshold on X number too, but alas that would be a pain I would imagine.

gschoenberger, 2019-01-10:
I've added regular expression support for excludes with commit 80a41ce60b4fd551d90fd942d7424041a1d386f7
Can you test it?
The first character of the exclude line hast to be '~' then you can use a regex, like:
~Memory Module . DDR4_P._A._ECC| Memory

gschoenberger, 2019-01-10:
Please take also commit 245d2e9e4a6d790523e802113cb4f0ef925f2329 into account.

tniedermeier, 2019-01-11:
Maybe I am doing something wrong...
String in my selexclude file:
~Intru|Physical Security
I understand that now all SEL entries which contain Intru are not printed, but they remain in the output.

Verbose output:

./check_ipmi_sensor -H ipmi-address -U adminuser -P adminpassword -L admin -vvv --sexclude selexclude 
------------- debug output for sel (-vvv is set): ------------
  /usr/sbin/ipmi-sel was executed with the following parameters:
    /usr/sbin/ipmi-sel -h ipmi-address -u adminuser -p adminpassword -l admin --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all --exclude-sensor-types=
  output of FreeIPMI:
ID   | Date        | Time     | Name                                              | Type              | State    | Event
1    | PostInit    | PostInit | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
2    | PostInit    | PostInit | Power Supply PS2 Status                           | Power Supply      | Critical | Power Supply Failure detected
3    | PostInit    | PostInit | Power Supply PS2 Status                           | Power Supply      | Critical | Power Supply Failure detected
4    | Jul-20-2017 | 10:29:49 | Power Supply PS2 Status                           | Power Supply      | Critical | Power Supply Failure detected
5    | Jul-20-2017 | 10:30:02 | Power Supply PS2 Status                           | Power Supply      | Critical | Power Supply Failure detected
6    | Jul-20-2017 | 10:37:20 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
7    | Aug-22-2017 | 15:10:15 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
8    | Sep-18-2017 | 16:10:52 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
9    | Sep-18-2017 | 16:11:08 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
10   | Sep-18-2017 | 16:35:02 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
11   | Sep-25-2017 | 10:43:30 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
12   | Sep-25-2017 | 10:43:47 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
13   | Sep-25-2017 | 11:20:30 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
14   | Sep-25-2017 | 11:22:19 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
15   | Sep-25-2017 | 12:12:31 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
16   | Dec-04-2017 | 10:50:19 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
17   | Dec-04-2017 | 10:50:46 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
18   | Dec-04-2017 | 11:51:46 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
19   | Dec-04-2017 | 11:16:44 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
20   | Dec-04-2017 | 11:42:49 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
21   | Dec-04-2017 | 13:37:45 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
22   | Dec-04-2017 | 14:19:37 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
23   | Dec-04-2017 | 14:29:15 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
24   | Dec-04-2017 | 14:53:51 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
25   | Dec-04-2017 | 15:15:46 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
26   | Dec-04-2017 | 15:19:58 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
27   | Dec-19-2017 | 11:05:34 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
28   | Dec-19-2017 | 12:05:42 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
29   | Dec-19-2017 | 12:08:07 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
30   | Dec-19-2017 | 12:34:08 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
31   | Dec-19-2017 | 12:36:02 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
32   | Dec-19-2017 | 13:28:16 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
33   | Jan-09-2018 | 11:16:21 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
34   | Jan-09-2018 | 11:16:49 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
[... cut ...]
ID   | Name            | Type              | State    | Reading    | Units | Lower NR   | Lower C    | Lower NC   | Upper NC   | Upper C    | Upper NR   | Event

--------------------- end of debug output ---------------------
-> Your server seems to be powered off. (Execution of FreeIPMI returned an empty output or only 1 header row!)
-> /usr/sbin/ipmi-sensors was executed with the following parameters:
   /usr/sbin/ipmi-sensors -h ipmi-address -u adminuser -p adminpassword -l admin --quiet-cache --sdr-cache-recreate --interpret-oem-data --output-sensor-state --ignore-not-available-sensors --driver-type=LAN_2_0 --output-sensor-thresholds

tniedermeier, 2019-01-11:
Sorry for my last reply, layer 8 problem...
If you utilize both new parameters selexclude and sexclude correctly, they do great.
I think we can close this issue.

closed 2019-01-11 by tniedermeier