thomas-krenn/check_ipmi_sensor_v3 #26: Power Consumption

MarcusCaepio, 2018-12-04:
Hi all,
on our supermicro server, we don't get the power consumption of the power supplies.
Is this not integrated? 

Regards,
Marcus

tk-wfischer, 2018-12-04:
Hi Marcus,

this is indeed currently not implemented. I assume that your Supermicro Server provides the data via the "Intel Node Manager" spec, which can be queried with FreeIPMI:
https://www.thomas-krenn.com/de/wiki/Intel_Node_Manager_Leistungsmessung_mit_FreeIPMI
https://www.thomas-krenn.com/en/wiki/Intel_Node_Manager_power_measurement_using_FreeIPMI

But currently this is not implemented in the plugin.

Regards,
Werner

veitw, 2018-12-04:
Hi Marcus,

we measure the power of SuperMicro servers via DCMI using this simple
script.

It does not implement any limit checks, as we regard this to be purely
informational, but feel free to add them. We query the value up to 3
times (see $MAX_TRIES) because on some machines, sometimes the BMC does
not determine the value on the first try.

You will need to add /usr/sbin/ipmi-dcmi to your sudoers for the user
the check runs as.

Best regards,
// Veit

---
#!/bin/bash

MAX_TRIES=3
RETRY_SLEEP=5
SUDO=/usr/bin/sudo

POWER=""
TRY=0
while [ -z "$POWER" -a $TRY -lt $MAX_TRIES ]; do
	TRY=$[$TRY +1]
	[ $TRY -gt 1 ] && sleep $RETRY_SLEEP
	POWER="$(LANG=C $SUDO /usr/sbin/ipmi-dcmi --get-system-power-statistics 2>/dev/null | /usr/bin/sed -n 's/^current power \+: \+\([0-9]\+\) watts$/\1/pi')"
done

if [ -z "$POWER" ]; then
	echo "UNKNOWN - Unable to determine power consumption."
	exit 3
fi

echo "OK - $POWER Watts | 'Power Consumption'=${POWER}W"
exit 0
---

tniedermeier, 2019-01-07:
Output of X9SCM-iiF with PMBus capable power supply:

Current Power                        : 106 Watts
Minimum Power over sampling duration : 80 watts
Maximum Power over sampling duration : 106 watts
Average Power over sampling duration : 100 watts
Time Stamp                           : 01/07/2019 - 15:01:34
Statistics reporting time period     : 14000 milliseconds
Power Measurement                    : Active

Output of a X11SSH-LN4F without PMBus capabilities:

Current Power                        : 0 Watts
Minimum Power over sampling duration : 0 watts
Maximum Power over sampling duration : 0 watts
Average Power over sampling duration : 0 watts
Time Stamp                           : 01/07/2019 - 14:49:44
Statistics reporting time period     : 5412000 milliseconds
Power Measurement                    : Not Available

gschoenberger, 2019-01-07:
Commit efcb343b698206f5dda1640e4a059f2742e089e5 uses ipmi-dcmi to fetch "Current Power" and adds it to performance data.
Anyone can test latest master?
THX

tniedermeier, 2019-01-08:
I tested it with two different systems, if I run the script with privilege level admin it seems fine.
The system is equipped with a X9SCM-iiF mainboard and PMBus capable power supplies.

./check_ipmi_sensor -H <HOST> -U <USER> -P <PASSWORD> -L admin
IPMI Status: Critical [33 system event log (SEL) entries present] | 'Current Power'=27 'System Temp'=27.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=31.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2625.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=975.00;600.00:18975.00;450.00:19050.00 'FAN A'=2700.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.81;0.54:1.49;0.51:1.52 '3.3VCC'=3.38;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.51;1.28:1.76;1.22:1.78 '5VCC'=5.02;4.58:5.34;4.32:5.60 '-12V'=-12.09;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.06;2.94:3.58;2.88:3.65 'VSB'=3.33;2.94:3.58;2.88:3.65 'AVCC'=3.39;2.94:3.58;2.88:3.65

Executed with user privileges:

./check_ipmi_sensor -H <HOST> -U <USER> -P <PASSWORD> -L user
Use of uninitialized value in string eq at ./check_ipmi_sensor line 406.
Use of uninitialized value in string eq at ./check_ipmi_sensor line 411.
IPMI Status: Can't use string ("") as a HASH ref while "strict refs" in use at ./check_ipmi_sensor line 945.

System with a X11SSH-LN4F and power supply without PMBus capabilities, admin privileges:

./check_ipmi_sensor -H <HOST> -U <USER> -P <PASSWORD> -L admin
IPMI Status: Can't use string ("") as a HASH ref while "strict refs" in use at ./check_ipmi_sensor line 945.

Plugin execution with user privileges on both systems result in this prompt:
./check_ipmi_sensor -H <HOST> -U <USER> -P <PASSWORD> -L user
Use of uninitialized value in string eq at ./check_ipmi_sensor line 406.
Use of uninitialized value in string eq at ./check_ipmi_sensor line 411.
IPMI Status: Can't use string ("") as a HASH ref while "strict refs" in use at ./check_ipmi_sensor line 945.

I think the problem with the user privilege is, that you need admin privileges to get power statistics.
ipmi-dcmi command executed with user privileges:
ipmi-dcmi -h <IP> -u testuser -p <PASSWORD> --get-system-power-statistics
ipmi-dcmi: password verification timeout

tniedermeier, 2019-01-10:
I tested the latest version again with the X11SSH-LN4F (no PMBus capabilities), admin privilege level seems fixed now:

-L admin

IPMI Status: Warning [3 system event log (SEL) entries present] | 'CPU Temp'=27.00;0.00:95.00;0.00:100.00 'PCH Temp'=27.00;16.00:90.00;5.00:95.00 'System Temp'=22.00;0.00:80.00;-5.00:85.00 'Peripheral Temp'=32.00;0.00:80.00;-5.00:85.00 'VcpuVRM Temp'=28.00;5.00:95.00;0.00:100.00 'DIMMA1 Temp'=24.00;5.00:80.00;0.00:85.00 'FAN1'=5600.00;700.00:25300.00;500.00:25400.00 'FAN2'=5600.00;700.00:25300.00;500.00:25400.00 '12V'=12.19;10.78:12.96;10.27:13.28 '5VCC'=4.97;4.48:5.39;4.30:5.55 '3.3VCC'=3.37;2.96:3.55;2.82:3.66 'VBAT'=2.94;2.58:3.48;2.50:3.59 'VCPU'=0.22;0.08:1.52;0.08:1.52 'VDIMMAB'=1.21;1.05:1.34;0.97:1.43 '5VSB'=4.92;4.48:5.39;4.38:5.55 '3.3VSB'=3.30;2.96:3.55;2.89:3.66 'VBMC 1.2V'=1.20;1.09:1.34;1.05:1.37 'VPCH 1.0V'=0.99;0.90:1.14;0.86:1.16

But if you run it with user privilege level two warnings/errors remain:

-L user

Use of uninitialized value in string eq at ./check_ipmi_sensor line 406.
Use of uninitialized value in string eq at ./check_ipmi_sensor line 411.
IPMI Status: Warning [3 system event log (SEL) entries present] | 'CPU Temp'=27.00;0.00:95.00;0.00:100.00 'PCH Temp'=27.00;16.00:90.00;5.00:95.00 'System Temp'=22.00;0.00:80.00;-5.00:85.00 'Peripheral Temp'=32.00;0.00:80.00;-5.00:85.00 'VcpuVRM Temp'=29.00;5.00:95.00;0.00:100.00 'DIMMA1 Temp'=24.00;5.00:80.00;0.00:85.00 'FAN1'=5500.00;700.00:25300.00;500.00:25400.00 'FAN2'=5600.00;700.00:25300.00;500.00:25400.00 '12V'=12.19;10.78:12.96;10.27:13.28 '5VCC'=4.95;4.48:5.39;4.30:5.55 '3.3VCC'=3.37;2.96:3.55;2.82:3.66 'VBAT'=2.94;2.58:3.48;2.50:3.59 'VCPU'=0.08;0.08:1.52;0.08:1.52 'VDIMMAB'=1.21;1.05:1.34;0.97:1.43 '5VSB'=4.92;4.48:5.39;4.38:5.55 '3.3VSB'=3.30;2.96:3.55;2.89:3.66 'VBMC 1.2V'=1.20;1.09:1.34;1.05:1.37 'VPCH 1.0V'=0.99;0.90:1.14;0.86:1.16

tniedermeier, 2019-01-10:
Unfortunately, the problems with the uninitialized values remain if executed with user privilege-level:

Use of uninitialized value in string eq at ./check_ipmi_sensor line 407.
Use of uninitialized value in string eq at ./check_ipmi_sensor line 412.
IPMI Status: Warning [3 system event log (SEL) entries present] | 'CPU Temp'=27.00;0.00:95.00;0.00:100.00 'PCH Temp'=28.00;16.00:90.00;5.00:95.00 'System Temp'=23.00;0.00:80.00;-5.00:85.00 'Peripheral Temp'=32.00;0.00:80.00;-5.00:85.00 'VcpuVRM Temp'=29.00;5.00:95.00;0.00:100.00 'DIMMA1 Temp'=24.00;5.00:80.00;0.00:85.00 'FAN1'=5600.00;700.00:25300.00;500.00:25400.00 'FAN2'=5600.00;700.00:25300.00;500.00:25400.00 '12V'=12.19;10.78:12.96;10.27:13.28 '5VCC'=4.95;4.48:5.39;4.30:5.55 '3.3VCC'=3.37;2.96:3.55;2.82:3.66 'VBAT'=2.92;2.58:3.48;2.50:3.59 'VCPU'=0.28;0.08:1.52;0.08:1.52 'VDIMMAB'=1.21;1.05:1.34;0.97:1.43 '5VSB'=4.92;4.48:5.39;4.38:5.55 '3.3VSB'=3.30;2.96:3.55;2.89:3.66 'VBMC 1.2V'=1.20;1.09:1.34;1.05:1.37 'VPCH 1.0V'=0.99;0.90:1.14;0.86:1.16

tniedermeier, 2019-01-10:
Here also the verbose output (-vvv) with user privilege-level:

------------- debug output for sel (-vvv is set): ------------
  /usr/sbin/ipmi-sel was executed with the following parameters:
    /usr/sbin/ipmi-sel -h ip-address -u username -p password -l user --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all --exclude-sensor-types=
  output of FreeIPMI:
ID   | Date        | Time     | Name                                             | Type              | State    | Event
1    | Nov-15-2018 | 08:51:44 | Sensor #255                                      | Session Audit     | Warning  | Invalid Username or Password
2    | Nov-15-2018 | 08:51:50 | Sensor #255                                      | Session Audit     | Warning  | Invalid Username or Password
3    | Nov-15-2018 | 08:51:52 | Sensor #255                                      | Session Audit     | Warning  | Invalid Username or Password
Use of uninitialized value in string eq at ./check_ipmi_sensor line 407.
Use of uninitialized value in string eq at ./check_ipmi_sensor line 412.
------------- debug output for sensors (-vvv is set): ------------
  script was executed with the following parameters:
    ./check_ipmi_sensor -H ip-address -U username -P password -L user -vvv
  check_ipmi_sensor version:
    3.12
  FreeIPMI version:
    ipmi-sensors - 1.4.11
  FreeIPMI was executed with the following parameters:
    /usr/sbin/ipmi-sensors -h ip-address -u username -p password -l user --quiet-cache --sdr-cache-recreate --interpret-oem-data --output-sensor-state --ignore-not-available-sensors --driver-type=LAN_2_0 --output-sensor-thresholds
  FreeIPMI return code: 0
  output of FreeIPMI:
ID   | Name            | Type              | State    | Reading    | Units | Lower NR   | Lower C    | Lower NC   | Upper NC   | Upper C    | Upper NR   | Event
4    | CPU Temp        | Temperature       | Nominal  | 27.00      | C     | 0.00       | 0.00       | 0.00       | 95.00      | 100.00     | 100.00     | 'OK'
71   | PCH Temp        | Temperature       | Nominal  | 27.00      | C     | 0.00       | 5.00       | 16.00      | 90.00      | 95.00      | 100.00     | 'OK'
138  | System Temp     | Temperature       | Nominal  | 23.00      | C     | -10.00     | -5.00      | 0.00       | 80.00      | 85.00      | 90.00      | 'OK'
205  | Peripheral Temp | Temperature       | Nominal  | 32.00      | C     | -10.00     | -5.00      | 0.00       | 80.00      | 85.00      | 90.00      | 'OK'
272  | VcpuVRM Temp    | Temperature       | Nominal  | 28.00      | C     | -5.00      | 0.00       | 5.00       | 95.00      | 100.00     | 105.00     | 'OK'
339  | DIMMA1 Temp     | Temperature       | Nominal  | 24.00      | C     | -5.00      | 0.00       | 5.00       | 80.00      | 85.00      | 90.00      | 'OK'
607  | FAN1            | Fan               | Nominal  | 5600.00    | RPM   | 300.00     | 500.00     | 700.00     | 25300.00   | 25400.00   | 25500.00   | 'OK'
674  | FAN2            | Fan               | Nominal  | 5600.00    | RPM   | 300.00     | 500.00     | 700.00     | 25300.00   | 25400.00   | 25500.00   | 'OK'
942  | 12V             | Voltage           | Nominal  | 12.19      | V     | 10.14      | 10.27      | 10.78      | 12.96      | 13.28      | 13.41      | 'OK'
1009 | 5VCC            | Voltage           | Nominal  | 4.97       | V     | 4.25       | 4.30       | 4.48       | 5.39       | 5.55       | 5.60       | 'OK'
1076 | 3.3VCC          | Voltage           | Nominal  | 3.37       | V     | 2.79       | 2.82       | 2.96       | 3.55       | 3.66       | 3.69       | 'OK'
1143 | VBAT            | Voltage           | Nominal  | 2.92       | V     | 2.38       | 2.50       | 2.58       | 3.48       | 3.59       | 3.67       | 'OK'
1210 | VCPU            | Voltage           | Nominal  | 0.80       | V     | 0.08       | 0.08       | 0.08       | 1.52       | 1.52       | 1.52       | 'OK'
1277 | VDIMMAB         | Voltage           | Nominal  | 1.21       | V     | 0.95       | 0.97       | 1.05       | 1.34       | 1.43       | 1.44       | 'OK'
1344 | 5VSB            | Voltage           | Nominal  | 4.92       | V     | 4.25       | 4.38       | 4.48       | 5.39       | 5.55       | 5.60       | 'OK'
1411 | 3.3VSB          | Voltage           | Nominal  | 3.30       | V     | 2.79       | 2.89       | 2.96       | 3.55       | 3.66       | 3.69       | 'OK'
1478 | VBMC 1.2V       | Voltage           | Nominal  | 1.20       | V     | 1.02       | 1.05       | 1.09       | 1.34       | 1.37       | 1.40       | 'OK'
1545 | VPCH 1.0V       | Voltage           | Nominal  | 0.99       | V     | 0.83       | 0.86       | 0.90       | 1.14       | 1.16       | 1.19       | 'OK'
1612 | Chassis Intru   | Physical Security | Nominal  | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'OK'

--------------------- end of debug output ---------------------
IPMI Status: Warning [3 system event log (SEL) entries present - details: (Sensor #255 = Warning, Session Audit, Invalid Username or Password), (Sensor #255 = Warning, Session Audit, Invalid Username or Password), (Sensor #255 = Warning, Session Audit, Invalid Username or Password) - fix the reported issues and clear your SEL or exclude specific SEL entries using the -sx or -xST option] | 'CPU Temp'=27.00;0.00:95.00;0.00:100.00 'PCH Temp'=27.00;16.00:90.00;5.00:95.00 'System Temp'=23.00;0.00:80.00;-5.00:85.00 [... cut ...]
Peripheral Temp = 32.00 (Status: Nominal)
VcpuVRM Temp = 28.00 (Status: Nominal)
DIMMA1 Temp = 24.00 (Status: Nominal)
FAN1 = 5600.00 (Status: Nominal)
FAN2 = 5600.00 (Status: Nominal)
12V = 12.19 (Status: Nominal)
5VCC = 4.97 (Status: Nominal)
3.3VCC = 3.37 (Status: Nominal)
VBAT = 2.92 (Status: Nominal)
VCPU = 0.80 (Status: Nominal)
VDIMMAB = 1.21 (Status: Nominal)
5VSB = 4.92 (Status: Nominal)
3.3VSB = 3.30 (Status: Nominal)
VBMC 1.2V = 1.20 (Status: Nominal)
VPCH 1.0V = 0.99 (Status: Nominal)
Chassis Intru = 'OK' (Status: Nominal)

gschoenberger, 2019-01-10:
Can you test commit 8b568a83af5257bf7939ffd8790e5e81296c7beb ?
THX, Georg

tniedermeier, 2019-01-11:
Now the X9SCM-iiF with PMBus power supply:
./check_ipmi_sensor -H ipmi-address -U admin -P adminpassword -L admin

IPMI Status: Critical [33 system event log (SEL) entries present] | 'Current Power'=25 'System Temp'=25.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=29.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=975.00;600.00:18975.00;450.00:19050.00 'FAN A'=2625.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.82;0.54:1.49;0.51:1.52 '3.3VCC'=3.38;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.51;1.28:1.76;1.22:1.78 '5VCC'=5.02;4.58:5.34;4.32:5.60 '-12V'=-12.09;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.06;2.94:3.58;2.88:3.65 'VSB'=3.33;2.94:3.58;2.88:3.65 'AVCC'=3.38;2.94:3.58;2.88:3.65

user privilege-level:

./check_ipmi_sensor -H ipmi-address -U user -P userpassword -L user

IPMI Status: Critical [33 system event log (SEL) entries present] | 'System Temp'=26.00;-5.00:80.00;-7.00:85.00 'Peripheral Temp'=30.00;-5.00:80.00;-7.00:85.00 'FAN 1'=2625.00;600.00:18975.00;450.00:19050.00 'FAN 3'=2700.00;600.00:18975.00;450.00:19050.00 'FAN 4'=975.00;600.00:18975.00;450.00:19050.00 'FAN A'=2625.00;600.00:18975.00;450.00:19050.00 'Vcore'=0.78;0.54:1.49;0.51:1.52 '3.3VCC'=3.38;2.94:3.58;2.88:3.65 '12V'=12.14;10.71:13.09;10.60:13.20 'VDIMM'=1.51;1.28:1.76;1.22:1.78 '5VCC'=5.02;4.58:5.34;4.32:5.60 '-12V'=-12.09;-13.46:-10.93;-13.65:-10.74 'VBAT'=3.06;2.94:3.58;2.88:3.65 'VSB'=3.33;2.94:3.58;2.88:3.65 'AVCC'=3.38;2.94:3.58;2.88:3.65

verbose modes

first admin:
------------- debug output for sel (-vvv is set): ------------
  /usr/sbin/ipmi-sel was executed with the following parameters:
    /usr/sbin/ipmi-sel -h address -u username -p password -l admin --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all --exclude-sensor-types=
  output of FreeIPMI:
ID   | Date        | Time     | Name                                              | Type              | State    | Event
1    | PostInit    | PostInit | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
2    | PostInit    | PostInit | Power Supply PS2 Status                           | Power Supply      | Critical | Power Supply Failure detected
3    | PostInit    | PostInit | Power Supply PS2 Status                           | Power Supply      | Critical | Power Supply Failure detected
4    | Jul-20-2017 | 10:29:49 | Power Supply PS2 Status                           | Power Supply      | Critical | Power Supply Failure detected
5    | Jul-20-2017 | 10:30:02 | Power Supply PS2 Status                           | Power Supply      | Critical | Power Supply Failure detected
6    | Jul-20-2017 | 10:37:20 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
7    | Aug-22-2017 | 15:10:15 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
8    | Sep-18-2017 | 16:10:52 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
9    | Sep-18-2017 | 16:11:08 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
10   | Sep-18-2017 | 16:35:02 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
11   | Sep-25-2017 | 10:43:30 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
12   | Sep-25-2017 | 10:43:47 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
13   | Sep-25-2017 | 11:20:30 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
14   | Sep-25-2017 | 11:22:19 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
15   | Sep-25-2017 | 12:12:31 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
16   | Dec-04-2017 | 10:50:19 | System Chassis Chassis Intru                      | Physical Security | Critical | General Chassis Intrusion
17   | Dec-04-2017 | 10:50:46 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
18   | Dec-04-2017 | 11:51:46 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
19   | Dec-04-2017 | 11:16:44 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
20   | Dec-04-2017 | 11:42:49 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
21   | Dec-04-2017 | 13:37:45 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
22   | Dec-04-2017 | 14:19:37 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
23   | Dec-04-2017 | 14:29:15 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
24   | Dec-04-2017 | 14:53:51 | Sensor #0                                         | OS Critical Stop  | Warning  | OS Graceful Shutdown
25   | Dec-04-2017 | 15:15:46 | Sensor #0                                         | OS Boot           | Nominal  | C: boot completed
[... cut ...]
FAN 1 = 2625.00 (Status: Nominal)
FAN 3 = 2700.00 (Status: Nominal)
FAN 4 = 975.00 (Status: Nominal)
FAN A = 2625.00 (Status: Nominal)
Vcore = 0.79 (Status: Nominal)
3.3VCC = 3.38 (Status: Nominal)
12V = 12.14 (Status: Nominal)
VDIMM = 1.51 (Status: Nominal)
5VCC = 5.02 (Status: Nominal)
-12V = -12.09 (Status: Nominal)
VBAT = 3.06 (Status: Nominal)
VSB = 3.33 (Status: Nominal)
AVCC = 3.38 (Status: Nominal)
Chassis Intru = 'OK' (Status: Nominal)
PS2 Status = 'Presence detected' (Status: Nominal)

gschoenberger, 2019-01-11:
OK, then I think we can close this issue...

MarcusCaepio, 2019-01-11:
Hi all thanks a lot,
am I wrong or is the Current Power missing in the Plugin output?
[image]
[image]

closed 2019-01-15 by gschoenberger