thomas-krenn/check_ipmi_sensor_v3 #30: How to handle "PSU Config | Version Change"

stephanhendl, 2019-08-21:
Hi,

one Fujitsu server shows at Id 78:

```
76  | Power Limit      | Power Unit                  | N/A      | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'Limit Not Exceeded'
77  | Power Unit       | Power Unit                  | Nominal  | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'Fully Redundant'
78  | PSU Config       | Version Change              | Warning  | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'Hardware change detected with associated Entity'
80  | PSU1             | OEM Reserved                | N/A      | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'OEM Event = 0002h'
81  | PSU2             | OEM Reserved                | N/A      | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'OEM Event = 0002h'
82  | Power Level      | OEM Reserved                | N/A      | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'OEM Event = 0008h'
83  | P-STATE Throttle | OEM Reserved                | N/A      | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'OEM Event = 0001h'
84  | System State     | System Event                | N/A      | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'OK'

```
I went through the iRMC but didn't find any place where I can commit something or whatever. Can I safely ignore that and disable it from checking?

Regards,
Stephan

mj84, 2020-06-22:
Hi, I just encountered the same issue on a Fujitsu as well.
Was able to ignore that warning by specifying `-xT Version_Change` as parameter.

Cheers,
Markus

nabil3399, 2024-01-26:
can you please tell me how and where to excute this command?

nabil3399, 2024-01-26:
> can you please tell me how and where to excute this command?

> Hi, I just encountered the same issue on a Fujitsu as well. Was able to ignore that warning by specifying `-xT Version_Change` as parameter.
> 
> Cheers, Markus

> can you please tell me how and where to excute this command?

gschoenberger, 2024-01-29:
You have to add this option to the plugin call, like:
$ check_ipmi_sensor -H 192.0.2.1 -U monitor -P monitor -L user -xT Version_Change

Then the sensor of IPMI type "Version Change" is excluded from the check!

nabil3399, 2024-01-29:
> You have to add this option to the plugin call, like: $ check_ipmi_sensor -H 192.0.2.1 -U monitor -P monitor -L user -xT Version_Change
> 
> Then the sensor of IPMI type "Version Change" is excluded from the check!

im so thankful for help , i need to do it using putty?

gschoenberger, 2024-01-30:
> > You have to add this option to the plugin call, like: $ check_ipmi_sensor -H 192.0.2.1 -U monitor -P monitor -L user -xT Version_Change
> > Then the sensor of IPMI type "Version Change" is excluded from the check!
> 
> im so thankful for help , i need to do it using putty?

You have to add this to the check_ipmi_sensor plugin call, wherever your are doing this!