thomas-krenn/check_ipmi_sensor_v3 #31: Incorrect --seloptions cmd-line argument passing to ipmi-sel

d4nys3k, 2019-09-16:
When multiple comma-separated options are passed to --seloptions, ipmi-sel execution fails due to `unrecognized option`.

Simple push of sel_options array to selcmd fixes this problem (see attached patch).
[check_ipmi_sensor.diff.gz](https://github.com/thomas-krenn/check_ipmi_sensor_v3/files/3616315/check_ipmi_sensor.diff.gz)

gschoenberger, 2019-09-17:
@tniedermeier @tk-wfischer Can you test the patch? If it works we can include it upstream...

tniedermeier, 2019-09-17:
Hello @d4nys3k,

thank you for your patch, I tested the patch and here are my results.
It seems that even the unpatched master branch handles mutiple parameters seperated by comma right.

Best regards,
Thomas

Output of the patched version:

$ ./check_ipmi_sensor -H <IPMI-BMC-IP> -U <admin-user> -P <admin-password> -L admin --seloptions '--date-range=09/01/2019-now,--assume-system-event-records,--oem-event-only' -vvv
------------- debug output for sel (-vvv is set): ------------
  /usr/sbin/ipmi-sel was executed with the following parameters:
    /usr/sbin/ipmi-sel -h <IPMI-BMC-IP> -u <admin-user> -p <admin-password> -l admin --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all --date-range=09/01/2019-now --assume-system-event-records --oem-event-only
  output of FreeIPMI:
------------- debug output for sensors (-vvv is set): ------------
  script was executed with the following parameters:
    ./check_ipmi_sensor -H <IPMI-BMC-IP> -U <admin-user> -P <admin-password> -L admin --seloptions --date-range=09/01/2019-now,--assume-system-event-records,--oem-event-only -vvv
  check_ipmi_sensor version:
    3.13
  FreeIPMI version:
    ipmi-sensors - 1.4.11
  FreeIPMI was executed with the following parameters:
    /usr/sbin/ipmi-sensors -h <IPMI-BMC-IP> -u <admin-user> -p <admin-password> -l admin --quiet-cache --sdr-cache-recreate --interpret-oem-data --output-sensor-state --ignore-not-available-sensors --driver-type=LAN_2_0 --output-sensor-thresholds
  FreeIPMI return code: 0
  output of FreeIPMI:
ID   | Name            | Type              | State    | Reading    | Units | Lower NR   | Lower C    | Lower NC   | Upper NC   | Upper C    | Upper NR   | Event
4    | CPU Temp        | Temperature       | Nominal  | 31.00      | C     | 0.00       | 0.00       | 0.00       | 95.00      | 100.00     | 100.00     | 'OK'
71   | PCH Temp        | Temperature       | Nominal  | 30.00      | C     | 0.00       | 5.00       | 16.00      | 90.00      | 95.00      | 100.00     | 'OK'
138  | System Temp     | Temperature       | Nominal  | 25.00      | C     | -10.00     | -5.00      | 0.00       | 80.00      | 85.00      | 90.00      | 'OK'
205  | Peripheral Temp | Temperature       | Nominal  | 34.00      | C     | -10.00     | -5.00      | 0.00       | 80.00      | 85.00      | 90.00      | 'OK'
272  | VcpuVRM Temp    | Temperature       | Nominal  | 35.00      | C     | -5.00      | 0.00       | 5.00       | 95.00      | 100.00     | 105.00     | 'OK'
339  | DIMMA1 Temp     | Temperature       | Nominal  | 26.00      | C     | -5.00      | 0.00       | 5.00       | 80.00      | 85.00      | 90.00      | 'OK'
607  | FAN1            | Fan               | Nominal  | 5700.00    | RPM   | 300.00     | 500.00     | 700.00     | 25300.00   | 25400.00   | 25500.00   | 'OK'
674  | FAN2            | Fan               | Nominal  | 5600.00    | RPM   | 300.00     | 500.00     | 700.00     | 25300.00   | 25400.00   | 25500.00   | 'OK'
942  | 12V             | Voltage           | Nominal  | 12.19      | V     | 10.14      | 10.27      | 10.78      | 12.96      | 13.28      | 13.41      | 'OK'
1009 | 5VCC            | Voltage           | Nominal  | 4.95       | V     | 4.25       | 4.30       | 4.48       | 5.39       | 5.55       | 5.60       | 'OK'
1076 | 3.3VCC          | Voltage           | Nominal  | 3.37       | V     | 2.79       | 2.82       | 2.96       | 3.55       | 3.66       | 3.69       | 'OK'
1143 | VBAT            | Voltage           | Nominal  | 2.97       | V     | 2.38       | 2.50       | 2.58       | 3.48       | 3.59       | 3.67       | 'OK'
1210 | VCPU            | Voltage           | Nominal  | 0.79       | V     | 0.08       | 0.08       | 0.08       | 1.52       | 1.52       | 1.52       | 'OK'
1277 | VDIMMAB         | Voltage           | Nominal  | 1.21       | V     | 0.95       | 0.97       | 1.05       | 1.34       | 1.43       | 1.44       | 'OK'
1344 | 5VSB            | Voltage           | Nominal  | 4.92       | V     | 4.25       | 4.38       | 4.48       | 5.39       | 5.55       | 5.60       | 'OK'
1411 | 3.3VSB          | Voltage           | Nominal  | 3.30       | V     | 2.79       | 2.89       | 2.96       | 3.55       | 3.66       | 3.69       | 'OK'
1478 | VBMC 1.2V       | Voltage           | Nominal  | 1.20       | V     | 1.02       | 1.05       | 1.09       | 1.34       | 1.37       | 1.40       | 'OK'
1545 | VPCH 1.0V       | Voltage           | Nominal  | 0.99       | V     | 0.83       | 0.86       | 0.90       | 1.14       | 1.16       | 1.19       | 'OK'
1612 | Chassis Intru   | Physical Security | Nominal  | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'OK'

--------------------- end of debug output ---------------------
IPMI Status: OK | 'CPU Temp'=31.00;0.00:95.00;0.00:100.00 'PCH Temp'=30.00;16.00:90.00;5.00:95.00 'System Temp'=25.00;0.00:80.00;-5.00:85.00 'Peripheral Temp'=34.00;0.00:80.00;-5.00:85.00 'VcpuVRM Temp'=35.00;5.00:95.00;0.00:100.00 'DIMMA1 Temp'=26.00;5.00:80.00;0.00:85.00 'FAN1'=5700.00;700.00:25300.00;500.00:25400.00 'FAN2'=5600.00;700.00:25300.00;500.00:25400.00 '12V'=12.19;10.78:12.96;10.27:13.28 '5VCC'=4.95;4.48:5.39;4.30:5.55 '3.3VCC'=3.37;2.96:3.55;2.82:3.66 'VBAT'=2.97;2.58:3.48;2.50:3.59 'VCPU'=0.79;0.08:1.52;0.08:1.52 'VDIMMAB'=1.21;1.05:1.34;0.97:1.43 '5VSB'=4.92;4.48:5.39;4.38:5.55 '3.3VSB'=3.30;2.96:3.55;2.89:3.66 'VBMC 1.2V'=1.20;1.09:1.34;1.0 [... cut ...]
Peripheral Temp = 34.00 (Status: Nominal)
VcpuVRM Temp = 36.00 (Status: Nominal)
DIMMA1 Temp = 26.00 (Status: Nominal)
FAN1 = 5600.00 (Status: Nominal)
FAN2 = 5700.00 (Status: Nominal)
12V = 12.13 (Status: Nominal)
5VCC = 4.95 (Status: Nominal)
3.3VCC = 3.37 (Status: Nominal)
VBAT = 2.97 (Status: Nominal)
VCPU = 1.02 (Status: Nominal)
VDIMMAB = 1.21 (Status: Nominal)
5VSB = 4.92 (Status: Nominal)
3.3VSB = 3.30 (Status: Nominal)
VBMC 1.2V = 1.20 (Status: Nominal)
VPCH 1.0V = 0.99 (Status: Nominal)
Chassis Intru = 'OK' (Status: Nominal)

d4nys3k, 2019-09-17:
Hello, I have troubles using that within bash script (at least version 4.2.46 used in Centos7), which I'm using as a simple wraper (for obtaining auth data from database).

When multiple comma-separated parameters are passed, ipmi-sel produces following error:
/usr/sbin/ipmi-sel: unrecognized option '--param1 --param2' (doesn't matter which parameter).

I tried to call check_ipmi_sensor is called in both ways:  _--sel-param --param1,--param2_ and  _--sel-param '--param1,--param2'_ (quoted like in your example) - both with similar result. This is not happening with patch I provided in this ticket. 
 
In your script I saw that your're handling _[freeipmi_options](https://github.com/thomas-krenn/check_ipmi_sensor_v3/blob/master/check_ipmi_sensor#L729)_ and _[sel_options](https://github.com/thomas-krenn/check_ipmi_sensor_v3/blob/master/check_ipmi_sensor#L355)_  variables quite differently at this time. I'm observing such problem in combination with --sel-only option (which I patched in #12) with event types limited by -ST parameter.

facundofc, 2021-03-04:
Hi there! I'm reviving this thread 'cause we're hitting this very same issue here.

I'd like to understand how you're running the `check_ipmi_sensor` script, because I'm seeing the same behavior as @d4nys3k, even using master:

```
$ ./check_ipmi_sensor-master -vvv --seloptions --date-range=09/01/2019-now,--assume-system-event-records,--oem-event-only
invalid range input

-> Execution of /usr/sbin/ipmi-sel failed with return code 1.
-> /usr/sbin/ipmi-sel was executed with the following parameters:
   sudo /usr/sbin/ipmi-sel --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all --date-range=09/01/2019-now --assume-system-event-records --oem-event-only
```

If I copy-paste and execute that exact `sudo` line right there, the command executes correctly. If I add quotes around all the options I included with `--seloptions`, I get the same error:

```
$ sudo /usr/sbin/ipmi-sel --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all '--date-range=09/01/2019-now --assume-system-event-records --oem-event-only'
invalid range input
```

This is relevant because I think this is how `check_ipmi_sensors` is executing (with the quotes).

How is it possible that it's working for you? Could you provide some insights?

Thanks!

afreiberger, 2021-03-05:
This must be down to different versions of IPC::Run package or behavior coded into the IPC::Run harness being used across different operating systems.  I believe that the standard of using separate args packed into the @command array is much more POSIX compliant for calls to exec at the end of the stream, otherwise, using spaces with a single arg requires that you are necessarily executing the command with a typical unix shell interpreter, but IPC::Run doesn't necessarily do that on all platforms.

To illustrate the point in a very simple fashion:

my @cmd = ("ls", "myfile1", "myfile2")
run \@cmd, ">", "outputfile"

Should behave much differently from:
my @cmd = ("ls", "myfile1 myfile2")
run \@cmd, ">", "outputfile"

I would expect two results in the first command and either no results or a single result of a third file named "myfile1 myfile2" with the space.

In the situation of the code at https://github.com/thomas-krenn/check_ipmi_sensor_v3/blob/master/check_ipmi_sensor#L368, you're creating the second case of making one big argv argument out of what should be separate argv arguments to the forked call.

afreiberger, 2021-03-05:
On drew@grimoire:~$ /usr/bin/perl
use IPC::Run qw( run ); #interact with processes
my @command = qw(ls);
push @command, "myfile1 myfile2";
run \@command, ">", "output";
/usr/bin/ls: cannot access 'myfile1 myfile2': No such file or directory

This is an example from Ubuntu Focal Fossa 20.04 with the following perl and IPC::Run versions.

ii  perl                                          5.30.0-9ubuntu0.2                     amd64        Larry Wall's Practical Extraction and Report Language
ii  libipc-run-perl                               20180523.0-2                          all          Perl module for running processes

gschoenberger, 2021-03-24:
THX for the pull request!

closed 2021-03-24 by gschoenberger