thomas-krenn/check_ipmi_sensor_v3 #33: IPMI PSU queries always show Critical

DebianGuru, 2020-04-07:
Thanks for your excellent work.  I've set this up for Nagios core but I seem to get a Critical error about the power supplies.   Here's what I get when I run it with the -vvv flags for debugging:

------------- debug output for sel (-vvv is set): ------------
  /usr/sbin/ipmi-sel was executed with the following parameters:
    /usr/sbin/ipmi-sel -h 172.17.8.91 --config-file /etc/ipmi-config/ipmi.cfg --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names
  output of FreeIPMI:
ID  | Date        | Time     | Name                                           | Type                        | State    | Event
1   | Feb-15-2020 | 09:03:20 | System Management Software SEL Fullness        | Event Logging Disabled      | Nominal  | Log Area Reset/Cleared
2   | Mar-23-2020 | 11:52:18 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
3   | Mar-23-2020 | 11:52:28 | Front Panel Board Front Panel                  | Entity Presence             | Nominal  | Entity Present
4   | Mar-23-2020 | 11:52:29 | Cable/Interconnect 2 Front Video               | Cable/Interconnect          | Nominal  | Cable/Interconnect is connected
5   | Mar-23-2020 | 11:52:38 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
6   | Mar-23-2020 | 11:52:44 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
7   | Mar-23-2020 | 11:52:57 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
8   | Mar-23-2020 | 11:53:02 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
9   | Mar-23-2020 | 11:54:23 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; User-initiated system setup
10  | Mar-23-2020 | 15:01:47 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
11  | Mar-23-2020 | 15:01:52 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
12  | Mar-23-2020 | 15:01:56 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
13  | Mar-23-2020 | 15:02:04 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
14  | Mar-23-2020 | 15:11:14 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Starting operating system boot process, e.g. calling Unsigned Int 19h
15  | Mar-23-2020 | 15:36:01 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
16  | Mar-23-2020 | 15:37:06 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Starting operating system boot process, e.g. calling Unsigned Int 19h
17  | Mar-24-2020 | 12:29:55 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
18  | Mar-24-2020 | 12:31:07 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Starting operating system boot process, e.g. calling Unsigned Int 19h
19  | Mar-24-2020 | 12:56:48 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
20  | Mar-24-2020 | 12:56:53 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
21  | Mar-24-2020 | 12:56:58 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
22  | Mar-24-2020 | 12:57:05 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
23  | Mar-24-2020 | 12:58:21 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Starting operating system boot process, e.g. calling Unsigned Int 19h
24  | Mar-24-2020 | 13:00:57 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
25  | Mar-24-2020 | 13:01:08 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
26  | Mar-24-2020 | 13:01:12 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
27  | Mar-24-2020 | 13:01:20 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
28  | Mar-24-2020 | 13:03:35 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Starting operating system boot process, e.g. calling Unsigned Int 19h
29  | Mar-24-2020 | 13:24:38 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
30  | Mar-24-2020 | 13:24:43 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
31  | Mar-24-2020 | 13:24:47 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
32  | Mar-24-2020 | 13:24:55 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
33  | Mar-24-2020 | 13:25:59 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; User-initiated system setup
34  | Mar-24-2020 | 13:27:53 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
35  | Mar-24-2020 | 13:29:18 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Starting operating system boot process, e.g. calling Unsigned Int 19h
36  | Mar-24-2020 | 13:45:04 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
37  | Mar-24-2020 | 13:46:14 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Starting operating system boot process, e.g. calling Unsigned Int 19h
38  | Mar-24-2020 | 13:50:34 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
39  | Mar-24-2020 | 13:52:49 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Starting operating system boot process, e.g. calling Unsigned Int 19h
40  | Mar-24-2020 | 14:03:18 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
41  | Mar-24-2020 | 14:04:30 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Starting operating system boot process, e.g. calling Unsigned Int 19h
42  | Mar-24-2020 | 14:53:35 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
43  | Mar-24-2020 | 14:54:46 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Starting operating system boot process, e.g. calling Unsigned Int 19h
44  | Mar-25-2020 | 13:35:44 | System Firmware Progress                       | System Firmware Progress    | Nominal  | System Firmware Progress ; Unspecified
45  | Mar-25-2020 | 13:36:53 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
46  | Mar-25-2020 | 13:36:59 | Power Unit 1 Host Power                        | Power Unit                  | Nominal  | Power Off/Power Down
[... cut ...]
 "Sensor Type(s) Power_Supply Status: Critical [Power Supply 2 = Critical (Power Supply), Power Unit 1 Power Resource = Critical (Power Unit), Power Supply 2 = Warning (Power Supply), Power Unit 1 Power Resource = Warning (Power Unit), Power Unit 1 Power Resource = Critical (Power Unit), Power Supply 1 = Critical (Power Supply), Power Supply 1 = Warning (Power Supply), Power Unit 1 Power Resource = Warning (Power Unit)]"

How are these in a Critical state if they returned "Nominal" result.  Maybe I don't understand or the script is looking for some other value to be returned?  Thanks for the help.

veitw, 2020-04-07:
Hi, 

this is caused by the power supply events in your system event log. This ensures that not only completely defective PSUs are detected, but also flaky ones, temporary problems or power circuit/UPS failures. Also other problems are detected this way that would not show up in interval checks, e.g. ECC corrections that might indicate upcoming RAM failures if recurring, or chassis events indicating physical attacks. 

You will want to review all events logged, maybe save one or more events for later (PSU failures, ECC errors,... as they indicate problems when recurring) and clear the log afterwards. Then the check will complete OK. 
You can use the ipmi-sel command for this.

Best regards, 
// Veit

DebianGuru, 2020-04-07:
>             <check_ipmi_sensor_v3@noreply.github.com>
>             <author@noreply.github.com>
>             show Critical (#33)

Thanks a million! It's a brand new server.  It was crying because at some
point during setup one PSU was not plugged in.

Jay Fluharty
Network Analyst
Wheeling-Nippon Steel Inc.
PO Box 635
Follansbee, WV 26037
jayf@wheeling-nipponsteel.com
1-304-527-4819

Scanned by IBM Email Security Management Services powered by
Symantec.Cloud. For more information please visit
http://www-935.ibm.com/services/us/index.wss/offerfamily/iss/a1026954

veitw, 2020-04-07:
Glad it helped. :) 

Please close the issue if the problem is solved for you, so the guys at TK can focus on something else. I am not affiliated with TK and thus cannot do this.

gschoenberger, 2020-04-08:
THX @veitw for clarification!
If you want to clear the IPMI System Event Log you can use `ipmi-sel --clear`.
On a healthy server normally no additional entries should appear in the event log.
All the best, Georg

closed 2020-04-08 by gschoenberger