thomas-krenn/check_ipmi_sensor_v3 #35: Password exposed in output

m4k5ym, 2020-04-16:
Hi,

is it possible to remove or censore the password in check output if something went wrong?

```
/usr/sbin/ipmi-sensors: username invalid

-> Execution of /usr/sbin/ipmi-sensors failed with return code 1.
-> /usr/sbin/ipmi-sensors was executed with the following parameters:
   /usr/sbin/ipmi-sensors -h 127.0.0.1 -u someuser -p somepassword -l user --quiet-cache --sdr-cache-recreate --interpret-oem-data --output-sensor-state --ignore-not-available-sensors --driver-type=LAN_2_0 --output-sensor-thresholds
```

IMHO this is not so good because normally every monitoring user is able to see the output with exposed password.

Sincerly

gschoenberger, 2020-04-20:
Hi!
Yes I think it should be possible. The long option is "--password" I guess?
- Georg

m4k5ym, 2021-07-12:
I'm out I stopped using Icinga 2 a while ago now.

Decide your self to keep it open or close it.

phibos, 2022-08-17:
This has been fixed with PR #50 I think this issue can be closed.

closed 2022-08-18 by gschoenberger