thomas-krenn/check_ipmi_sensor_v3 #36: Should not warn when CPU2 is in state 'Not Installed'

taupehat, 2020-06-23:
This leads to a bogus CPU temperature warning on multi-socket systems with one socket unpopulated:

1210 | CPU1 Temp     | OEM Reserved      | Nominal  | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'Low'
1277 | CPU2 Temp     | OEM Reserved      | Warning  | N/A        | N/A   | N/A        | N/A        | N/A        | N/A        | N/A        | N/A        | 'Not Installed'

Best to ignore this...

taupehat, 2020-06-24:
Here's my cheesed-in fix - probably much better ways but this works:
```
--- check_ipmi_sensor   2020-06-24 09:46:49.913513616 -0700
+++ check_ipmi_sensor.latest    2020-06-24 09:44:49.792504662 -0700
@@ -922,9 +922,6 @@
                                        if(exists $row->{'event'} && ($row->{'event'} =~ /\'.*((Device|Entity) (Absent|Removed)).*\'/)){
                                                $check_sensor_state = 0;
                                        }
-                                        if(exists $row->{'event'} && ($row->{'event'} =~ /\'.*(Not Installed).*\'/)){
-                                                $check_sensor_state = 0;
-                                        }
                                        if(exists $row->{'reading'} && ($row->{'reading'} =~ /\'.*((Device|Entity) (Absent|Removed)).*\'/)){
                                                $check_sensor_state = 0;
                                        }
```

tniedermeier, 2020-07-03:
Hi @taupehat,

thanks for your code, I'll discuss it with @gschoenberger and we'll keep you updated.

Best regards,
Thomas