thomas-krenn/check_ipmi_sensor_v3 #39: Unknown SEL records breaks SEL parsing

mididix, 2020-10-08:
Hello,

On Gigabyte BMCs, we sometime get unsupported records like this :

``` 
[root@gmonitor5 ~]$ ipmi-sel -h bmc5 -u xxx -p xxx
Caching SDR repository information: /root/.freeipmi/sdr-cache/sdr-cache-gmonitor5.bmc5
Caching SDR record 68 of 68 (current record ID 68) 
ID | Date        | Time     | Name           | Type                     | Event
1  | Jun-30-2020 | 12:27:19 | SEL            | Event Logging Disabled   | Log Area Reset/Cleared
Unknown SEL Record Type: 4h
3  | Aug-04-2020 | 06:46:12 | Sensor #0      | System Event             | OEM System Boot Event
4  | Aug-04-2020 | 06:46:53 | Sensor #0      | System Event             | OEM System Boot Event
5  | Aug-04-2020 | 06:48:40 | Sensor #34     | OS Boot                  | boot completed - boot device not specified
```

We use freeipmi-1.5.7-3.el7.x86_64 from RHEL 7.8, and this output breaks down check_ipmi SEL parsing with this output:

``` bash
[belenos root@gmonitor5 ~]$ ./check_ipmi_sensor -H bmc5 -U xxx -P xxx -L admin
Use of uninitialized value $name in string eq at ./check_ipmi_sensor line 486, <FH> line 7.
Use of uninitialized value $name in string eq at ./check_ipmi_sensor line 486, <FH> line 7.
Use of uninitialized value $name in string eq at ./check_ipmi_sensor line 486, <FH> line 7.
Use of uninitialized value $name in string eq at ./check_ipmi_sensor line 486, <FH> line 7.
Use of uninitialized value $name in string eq at ./check_ipmi_sensor line 486, <FH> line 7.
Use of uninitialized value $name in string eq at ./check_ipmi_sensor line 486, <FH> line 7.
Use of uninitialized value $name in string eq at ./check_ipmi_sensor line 486, <FH> line 7.
IPMI Status: Use of uninitialized value in string ne at ./check_ipmi_sensor line 994.
Use of uninitialized value in string ne at ./check_ipmi_sensor line 997.
Critical [1 system event log (SEL) entry present] | 'Current Power'=276 'CPU0_PWR'=84.00 'CPU1_PWR'=78.00 'CPU0_TEMP'=50.00;5.00:97.00;0.00:100.00 'CPU1_TEMP'=47.00;5.00:97.00;0.00:100.00 'DIMMG0_TEMP'=43.00;5.00:84.00;0.00:87.00 'DIMMG1_TEMP'=42.00;5.00:84.00;0.00:87.00 'DIMMG2_TEMP'=33.00;5.00:84.00;0.00:87.00 'DIMMG3_TEMP'=35.00;5.00:84.00;0.00:87.00 'CPU0_DTS'=50.00 'CPU1_DTS'=53.00 'M2_G0_AMB_TEMP'=42.00;~:92.00;~:96.00 'MB_TEMP1'=45.00;5.00:85.00;0.00:90.00 'MB_TEMP2'=45.00;5.00:85.00;0.00:90.00 'P_12V'=11.89;10.79:13.13;10.27:13.65 'P_1V0_AUX_LAN'=0.98;0.90:1.11;0.85:1.16 'P0_VDD_18'=1.80;1.63:1.98;1.55:2.06 'P1_VDD_18'=1.81;1.63:1.98;1.55:2.06 'P_3V3'=3.36;2.96:3.62;2.82:3.75 'P_5V'=5.04;4.50:5.47;4.29:5.68 'P_5V_STBY'=5.06;4.50:5.47;4.29:5.68 'P_VBAT'=3.14;2.69:;2.54: 'P0_VDDCR_SOC'=0.73;0.45:1.40;0.40:1.46 'P0_VDDCR_CPU'=0.53;0.45:1.40;0.40:1.46 'P1_VDDCR_CPU'=0.53;0.45:1.40;0.40:1.46 'P1_VDDCR_SOC'=0.71;0.45:1.40;0.40:1.46 'P0_VDDIO_ABCD'=1.22;1.07:1.32;1.02:1.36 'P0_VDDIO_EFGH'=1.23;1.07:1.32;1.02:1.36 'P1_VDDIO_ABCD'=1.21;1.07:1.32;1.02:1.36 'P1_VDDIO_EFGH'=1.20;1.07:1.32;1.02:1.36 'VR_P0_IOUT'=1.00;~:228.00;~:248.00 'VR_P1_IOUT'=2.00;~:228.00;~:248.00 'VR_DIMMG0_IOUT'=28.00;~:228.00;~:248.00 'VR_DIMMG1_IOUT'=26.00;~:228.00;~:248.00 'VR_DIMMG2_IOUT'=27.00;~:228.00;~:248.00 'VR_DIMMG3_IOUT'=27.00;~:228.00;~:248.00 'VR_P0_TEMP'=35.00;5.00:115.00;0.00:120.00 'VR_P1_TEMP'=35.00;5.00:115.00;0.00:120.00 'VR_DIMMG0_TEMP'=53.00;5.00:115.00;0.00:120.00 'VR_DIMMG1_TEMP'=50.00;5.00:115.00;0.00:120.00 'VR_DIMMG2_TEMP'=32.00;5.00:115.00;0.00:120.00 'VR_DIMMG3_TEMP'=31.00;5.00:115.00;0.00:120.00 'VR_P0_VIN'=12.00;10.75:13.25;10.38:13.62 'VR_P1_VIN'=12.12;10.75:13.25;10.38:13.62 'VR_DIMMG0_VIN'=11.87;10.75:13.25;10.38:13.62 'VR_DIMMG1_VIN'=12.00;10.75:13.25;10.38:13.62 'VR_DIMMG2_VIN'=11.87;10.75:13.25;10.38:13.62 'VR_DIMMG3_VIN'=11.87;10.75:13.25;10.38:13.62 'VR_P0_VOUT'=0.75;0.45:1.41;0.40:1.46 'VR_P1_VOUT'=0.75;0.45:1.41;0.40:1.46 'VR_DIMMG0_VOUT'=1.23;1.09:1.33;1.02:1.39 'VR_DIMMG1_VOUT'=1.25;1.09:1.33;1.02:1.39 'VR_DIMMG2_VOUT'=1.25;1.09:1.33;1.02:1.39 'VR_DIMMG3_VOUT'=1.25;1.09:1.33;1.02:1.39
```

gschoenberger, 2020-10-28:
Can you test latest commit?

mididix, 2020-10-28:
Yup, it works for me, thanks 👍

commit 33cc88e9 (2020-10-28): Remove string "Unknown SEL Record Type" from SEL output - fixes #39
closed 2020-10-28 by mididix