thomas-krenn/check_ipmi_sensor_v3 #41: Optional additional FRU information in output

MarcusCaepio, 2020-10-30:
Good morning y'all,
Beside the serial number, we are also interested in the Asset Tag and Board Information in the output.
I am currently working on "--assettag" and "--board" options to display them. My idea is, that "--fru" is mandatory for both new options. This way, the script just uses the already existent "@fruoutput" and the code just has to be changed a little bit. 

Are you also interested in these options? If yes, I would create a PR, when it's finished.
Cheers and have a nice weekend,
Marcus

gschoenberger, 2020-10-30:
Yes! Please create a PR and I will review/merge it,

TXH - Georg

MarcusCaepio, 2020-10-30:
Hi Georg,
PR is done. If you merge it, I will re-fork the [Icinga2](https://github.com/Icinga/icinga2) Repo, adjust the ITL command and create a PR for it to have the current version also running with the next release.

Cheers,
Marcus

commit 5e4e2498 (2020-10-31): Resolves #41

Added the options --assettag and --board for further information, when --fru is used
commit 77a2d276 (2020-10-31): Merge pull request #42 from MarcusCaepio/master

Resolves #41
closed 2020-10-31 by gschoenberger