thomas-krenn/check_ipmi_sensor_v3 #44: k_g invalid

Napsty, 2020-12-16:
Tried to use check_ipmi_sensor v3 with a Cisco UCS C220 M5 server but it seems that some parameter for ipmi-sel is missing:

```
# ./check_ipmi_sensor -H CIMCIP -U admin -P secret -L admin -vvv
/usr/sbin/ipmi-sel: k_g invalid

-> Execution of /usr/sbin/ipmi-sel failed with return code 1.
-> /usr/sbin/ipmi-sel was executed with the following parameters:
   /usr/sbin/ipmi-sel -h CIMCIP -u admin -p secret -l admin --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all
```

According to manpage, the k_g is:

> Specify the K_g BMC key to use when authenticating with the remote host for IPMI 2.0. If not specified, a null key is assumed. To input the key in hexadecimal form, prefix the string with '0x'. E.g., the key 'abc' can be entered with the either the string 'abc' or the string '0x616263' 

What exactly is expected here? Are there (additional) IPMI configurations necessary on UCS side? The options are rather basic, basically only allows to enable IPMI and set a role - that's it.

Any ideas?

gschoenberger, 2020-12-18:
I am afraid I cannot provide any support in this case. Maybe @tk-wfischer or @tniedermeier can?

urfin78, 2021-06-25:
I think I stumbled upon the same error. For me the solution was to provide the key as options for both **ipmi-sel** with the `--seloptions` parameter and for **ipmi-sensors** with the `-O` parameter like
```bash
# ./check_ipmi_sensor -H CIMCIP -U admin -P secret -L admin -O --k-g=0x12345678 --seloptions --k-g=0x12345678 -vvv
```

gschoenberger, 2021-06-28:
@Napsty Can you test the given options from @urfin78 ?

idandr, 2024-09-24:
Had the same issue here ("k_g invalid").

Thanks a lot @urfin78 for the solution. As expected from your description, adding `-O --k-g=0x12345678` is sufficient when using `--nosel`.

gschoenberger, 2024-09-25:
Thanks for confirming a working solution!

closed 2024-09-25 by gschoenberger