thomas-krenn/check_ipmi_sensor_v3 #45: IPMI account/password got displayed when command failed

tjyang, 2021-02-06:
WHAT:  Security risk of exposing IMM/iDRAC's credential when underneath ipmi commands failed. 
WHY  :  $returncode !=0 will display IPMI account and password from command output and on web page.
HOW: 
- Adding one extra option like --oncmdout to not displaying command output so one can ensure no password when in production.
- I can submit a PR if there is interest.

selcukKaraca, 2021-07-13:
Hi
you should use "-f server.cfg"
in the file, you should specify the following:
# cat server.cfg 
username nagios
password abcd...!
privilege-level operator
#

tjyang, 2021-07-13:
@selcukKaraca , Thanks for the suggestion. But "-f" is more work(IMHO) given that this will generate more "server.cfg" files to maintain with, for site that hosts with much different accounts/passwords.

selcukKaraca, 2021-07-13:
as a solution I have implemented a wrapper script. I use nagios for monitoring. there is a table (ipmi_table.cfg) which includes the following fields
OS_IP ILO_IP AUTH_FILE OPTIONS

my wrapper script looks at this table, find necessary arguments and construct perfect check_ipmi_sensors command.
like this one:

```
# cat check_ipmi_wrapper.sh 
#this script constructs check-ipmi-sensor command using hostname parameter.
#it does this by looking up from a table ipmi_table.cfg
# written by mehmet selcuk karaca

FOUND=FALSE
IPMI_TABLE=/path/to/ipmi_table.cfg
HOST_IP=$2
#nagios unknown exit value
UNKNOWN=3

while read LINE
do
  IP=$(echo $LINE | cut -f 1 -d " ")
  #we have found IP (given as parameter) in the table. Now get other required arguments
  if [ $IP = $HOST_IP ]; then
     ILO=$(echo $LINE | cut -f 2 -d " ")
     AUTH_FILE=$(echo $LINE | cut -f 3 -d " ")
     AUTH_FILE=/path/to/$AUTH_FILE
     OPTIONS=$(echo $LINE | cut -f 4- -d " ")
     FOUND=TRUE
     break
  fi
done <$IPMI_TABLE

if [ $FOUND = "TRUE" ]; then
    /path/to/check_ipmi_sensor -H $ILO -f $AUTH_FILE $OPTIONS -v
else
   echo "$HOST_IP could not be found in $IPMI_TABLE please add IP and related info to the table."
   exit $UNKNOWN
fi
```

tjyang, 2021-07-13:
@selcukKaraca , are you interested to see how I resolve this issue by modifying perl code directly by adding --nocmdout argument  ?