thomas-krenn/check_ipmi_sensor_v3 #48: Use of uninitialized value

ef-tiramisu, 2021-07-20:
Hi,

Trying to get the latest version of the script to run on our Icinga System.

We get the following errors when executing the script, followed by the output of the IPMI values, however the script does not exit cleanly causing an issue within Icinga.

```
Use of uninitialized value in concatenation (.) or string at /usr/lib/nagios/plugins/check_ipmi_sensor3 line 973.
Use of uninitialized value in concatenation (.) or string at /usr/lib/nagios/plugins/check_ipmi_sensor3 line 973.
Use of uninitialized value in string ne at /usr/lib/nagios/plugins/check_ipmi_sensor3 line 984.
Use of uninitialized value in string ne at /usr/lib/nagios/plugins/check_ipmi_sensor3 line 984.
Use of uninitialized value in concatenation (.) or string at /usr/lib/nagios/plugins/check_ipmi_sensor3 line 985.
Use of uninitialized value in concatenation (.) or string at /usr/lib/nagios/plugins/check_ipmi_sensor3 line 985.
Use of uninitialized value in string ne at /usr/lib/nagios/plugins/check_ipmi_sensor3 line 994.
Use of uninitialized value in string ne at /usr/lib/nagios/plugins/check_ipmi_sensor3 line 972.
Use of uninitialized value in string ne at /usr/lib/nagios/plugins/check_ipmi_sensor3 line 972.
```
etc.

We are using:

- FreeIPMI 1.4.11
- Perl 5.22.1

Thanks in advance for any help.

gschoenberger, 2021-07-21:
Can you post the complete plugin output with "-vvv"? (excluding passwords if any are present in the output)

ef-tiramisu, 2021-07-21:
Here is a copy of the full Output with -vvv

[check_ipmi_sensor_output.txt](https://github.com/thomas-krenn/check_ipmi_sensor_v3/files/6854120/check_ipmi_sensor_output.txt)