thomas-krenn/check_ipmi_sensor_v3 #54: VBAT issue

michaelschefczyk, 2022-05-02:
Dear All,

I use your scripts v3 and v2 to check IPMI for my Supermicro servers. First of all: Thanks for providing them! Recently, I did acquire two X12DPL-NT6 motherboards. For one of them, both versions of the script report "IPMI Status: Critical [VBAT = Critical]“ when seen via nagios. Initially BMC also complains about one fan spinning too slow. While that can be cured with something like "ipmitool -H xxx -U xxx -P xxx sensor thresh "FANA" lcr 560", I did not find any cure against the VBAT issue. Both servers just report that the VBAT is present. Both are new and seem to have no voltage problems.

The BMC gui indicates nothing about the VBAT issues. A misconfiguration is unlikely: If is assign the BMC IP address in nagios to a different server, the VBAT issue travels with the IP. Other issues did not occur.

Command line output is similar for both systems 192.168.1.74 and 192.168.1.75. Command line output is IPMI Status: OK for both systems, just via nagios, the VBAT-issue does come up for system 192.168.1.75:

[root@gatewayIs10 plugins]# ./check_ipmi_sensor_v3 -H 192.168.1.74 -f /etc/freeipmi/freeipmi.conf -nosel
IPMI Status: OK | 'CPU1 Temp'=45.00;;5.00:92.00 'CPU2 Temp'=45.00;;5.00:92.00 'PCH Temp'=37.00;;5.00:90.00 'System Temp'=48.00;;5.00:85.00 'Peripheral Temp'=39.00;;5.00:85.00 'NIC Temp'=64.00;;5.00:100.00 'CPU1_VRMIN Temp'=61.00;;5.00:100.00 'CPU1_VRMIO Temp'=60.00;;5.00:100.00 'CPU2_VRMIN Temp'=64.00;;5.00:100.00 'CPU2_VRMIO Temp'=63.00;;5.00:100.00 'P1_VRMAC Temp'=54.00;;5.00:100.00 'P1_VRMEG Temp'=56.00;;5.00:100.00 'P2_VRMAC Temp'=61.00;;5.00:100.00 'P2_VRMEG Temp'=55.00;;5.00:100.00 'FAN1'=980.00;;700.00:35560.00 'FAN2'=980.00;;700.00:35560.00 'FAN3'=980.00;;700.00:35560.00 'FAN4'=980.00;;700.00:35560.00 'FAN5'=840.00;;700.00:35560.00 'FANA'=700.00;;560.00:35560.00 'P1_DIMMAC Temp'=46.00;;5.00:85.00 'P1_DIMMEG Temp'=52.00;;5.00:85.00 'P2_DIMMAC Temp'=54.00;;5.00:85.00 'P2_DIMMEG Temp'=52.00;;5.00:85.00 '12V'=11.83;;10.51:13.25 '5VCC'=4.96;;4.38:5.53 '3.3VCC'=3.28;;2.89:3.64 'Vcpu1VCCIN'=1.77;;1.58:2.02 'Vcpu1VCCIO'=1.00;;0.85:1.17 'Vcpu2VCCIN'=1.77;;1.58:2.02 'Vcpu2VCCIO'=1.00;;0.85:1.17 'Vp1AC'=1.22;;1.05:1.37 'Vp1EG'=1.22;;1.05:1.37 'Vp2AC'=1.22;;1.05:1.37 'Vp2EG'=1.22;;1.05:1.37 '5VSB'=5.04;;4.37:5.55 '3.3VSB'=3.30;;2.89:3.64 '2.5V BMC'=2.53;;2.26:2.85 '1.8V BMC'=1.81;;1.60:2.04 '1.2V BMC'=1.18;;1.05:1.34 '1.0V BMC'=0.99;;0.88:1.13 '1.8V PCH'=1.80;;1.58:2.00 'PVNN PCH'=0.99;;0.87:1.11 '1.05V PCH'=1.05;;0.93:1.19

[root@gatewayIs10 plugins]# ./check_ipmi_sensor_v3 -H 192.168.1.75 -f /etc/freeipmi/freeipmi.conf -nosel
IPMI Status: OK | 'CPU1 Temp'=44.00;;5.00:92.00 'CPU2 Temp'=50.00;;5.00:92.00 'PCH Temp'=41.00;;5.00:90.00 'System Temp'=50.00;;5.00:85.00 'Peripheral Temp'=37.00;;5.00:85.00 'NIC Temp'=68.00;;5.00:100.00 'CPU1_VRMIN Temp'=63.00;;5.00:100.00 'CPU1_VRMIO Temp'=60.00;;5.00:100.00 'CPU2_VRMIN Temp'=66.00;;5.00:100.00 'CPU2_VRMIO Temp'=59.00;;5.00:100.00 'P1_VRMAC Temp'=54.00;;5.00:100.00 'P1_VRMEG Temp'=57.00;;5.00:100.00 'P2_VRMAC Temp'=64.00;;5.00:100.00 'P2_VRMEG Temp'=55.00;;5.00:100.00 'FAN1'=1260.00;;700.00:35560.00 'FAN2'=1260.00;;700.00:35560.00 'FAN3'=1120.00;;700.00:35560.00 'FAN4'=1260.00;;700.00:35560.00 'FAN5'=980.00;;700.00:35560.00 'FANA'=840.00;;560.00:35560.00 'P1_DIMMAC Temp'=46.00;;5.00:85.00 'P1_DIMMEG Temp'=51.00;;5.00:85.00 'P2_DIMMAC Temp'=53.00;;5.00:85.00 'P2_DIMMEG Temp'=52.00;;5.00:85.00 '12V'=11.83;;10.51:13.25 '5VCC'=4.92;;4.38:5.53 '3.3VCC'=3.30;;2.89:3.64 'Vcpu1VCCIN'=1.76;;1.58:2.02 'Vcpu1VCCIO'=1.00;;0.85:1.17 'Vcpu2VCCIN'=1.75;;1.58:2.02 'Vcpu2VCCIO'=1.00;;0.85:1.17 'Vp1AC'=1.22;;1.05:1.37 'Vp1EG'=1.22;;1.05:1.37 'Vp2AC'=1.22;;1.05:1.37 'Vp2EG'=1.22;;1.05:1.37 '5VSB'=5.04;;4.37:5.55 '3.3VSB'=3.32;;2.89:3.64 '2.5V BMC'=2.51;;2.26:2.85 '1.8V BMC'=1.79;;1.60:2.04 '1.2V BMC'=1.17;;1.05:1.34 '1.0V BMC'=0.99;;0.88:1.13 '1.8V PCH'=1.81;;1.58:2.00 'PVNN PCH'=1.01;;0.87:1.11 '1.05V PCH'=1.04;;0.93:1.19

Does this ring a bell with anyone?

Thanks!

Michael Schefczyk

tk-wfischer, 2022-05-03:
Hi Michael,

thank you for using the plugin.
* _For one of them, both versions of the script report "IPMI Status: Critical [VBAT = Critical]“ **when seen via nagios**._
* _The BMC gui indicates nothing about the VBAT issues. A misconfiguration is unlikely: If is assign the BMC IP address in nagios to a different server, the VBAT issue travels with the IP. Other issues did not occur._
* _Command line output is similar for both systems 192.168.1.74 and 192.168.1.75. **Command line output is IPMI Status: OK for both systems**, just via nagios, the VBAT-issue does come up for system 192.168.1.75_

As you see the "IPMI Status: Critical [VBAT = Critical]" only via Nagios and not when you execute the plugin in the command line, I think that you may have a caching issue in Nagios.

For further investigation, please execute the plugin on the command line as user "nagios". If the issue is then also on the command line present, delete the SDR cache (as user "nagios") - see http://www.gnu.org/software/freeipmi/manpages/man8/ipmi-sensors-config.8.html (--flush-cache option)

Best regards,
Werner

michaelschefczyk, 2022-05-03:
Dear Werner,

Thank you very much!

Indeed, running the command as user nagios does show the [VBAT = Critical] issue. Running the command as root does show everything as OK.

What I did not understand from the man page is, how to clear the SDR cache. It seems that some parameters need to be set in addition to the -f

-bash-4.2$ ipmi-sensors-config -f
/usr/sbin/ipmi-config: option requires an argument -- 'f'

Experimenting with host user and parameter does not look like it would work. Could you please help me with this hurdle?

Regards,

Michael

michaelschefczyk, 2022-05-04:
For everyone in a similar situation:

A likely location for the SDR cache files is: /var/spool/nagios/.freeipmi/sdf-cache

Deleting the file corresponding to the IP address of the IPMI device showing the problem does solve the issue. Removing freeipmi and installing it again, does not solve the issue.

tk-wfischer, 2022-05-05:
Hi Michael,

thank you for your feedback and sorry that I haven't been able to answer sooner.
I'm happy that you have been able to fix your issue.

I'll hereby close this issue. In case that you have any further questions, just open another issue.

Best regards and have fun with the plugin,
Werner

closed 2022-05-05 by tk-wfischer