thomas-krenn/check_ipmi_sensor_v3 #56: Flag to add additional options for FreeIPMI doesn't apply for SEL checks

knbrende, 2022-10-27:
To access some of our servers, we have to set a workaround flag.
While this works great for sensors through the "-O" flag, it doesn't apply to SEL checks.

```
/usr/lib/nagios/plugins/check_ipmi_sensor -H <hostname> -U <username> -P <password> -L user --selonly -O '-W authcap'
/usr/sbin/ipmi-sel: username invalid

-> Execution of /usr/sbin/ipmi-sel failed with return code 1.
-> /usr/sbin/ipmi-sel was executed with the following parameters:
   /usr/sbin/ipmi-sel -h <hostname> -u <username> -p <password> -l user --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all --exclude-sensor-types=
```

I'm using the latest version 3.13

knbrende, 2022-11-02:
I just realized version 3.13 on master branch is not the same as tagged version 3.13.

May I ask you to bump the version to 3.14 and create a new tag for it as our rpm packaging heavily relies on Github tags?

gschoenberger, 2022-11-02:
Bumped 3.14!
https://github.com/thomas-krenn/check_ipmi_sensor_v3/releases/tag/v3.14

closed 2022-11-03 by knbrende