thomas-krenn/check_ipmi_sensor_v3 #57: How to for Icinga director

nobe80, 2022-12-08:
Hi togehter,

is there anywhere a how to for configuring in Icinga2 director? I installed according to https://www.thomas-krenn.com/en/wiki/IPMI_Sensor_Monitoring_Plugin_setup

but get always an usage error:
Plugin Output
Error: -f or -U -P -L missing.Usage:
check_ipmi_sensor -H [-f | -U -P -L ] [-O ] [-b] [-T ] [-ST ] [-x ] [-xT ] [-xST ] [-o zenoss] [-D ] [-h] [-V] [-fc ] [--fru] [--assettag] [--board] [--nosel] [--selonly] [--seltail ] [-sx|--selexclude ] [-xx|--sexclude ] [-us|--unify-sensors ] [--nosudo [--nothresholds] [--noentityabsent] [-s ] [-h] [-V] [-v|-vv|-vvv]

or with BMC Username & Password:

Plugin Output
/usr/sbin/ipmi-sel: permission denied
-> Execution of /usr/sbin/ipmi-sel failed with return code 1.
-> /usr/sbin/ipmi-sel was executed with the following parameters:
   /usr/sbin/ipmi-sel -h 127.0.0.1 -u admin -p ypvnIXXX -l user --driver-type=LAN_2_0 --output-event-state --interpret-oem-data --entity-sensor-names --sensor-types=all

What i am doing wrong?

regards

gschoenberger, 2022-12-12:
You are getting: "/usr/sbin/ipmi-sel: permission denied"
Can you run the plugin call as nagios/icinga user on the CLI? Maybe the execution bit is missing?

mwaldmueller, 2022-12-12:
The user doesn't have enough permissions to execute the command, you can set the permissions in `/etc/sudoers` e.g. for user "icinga" like this:

```
icinga ALL=(ALL)NOPASSWD:/usr/sbin/ipmi-sensors
icinga ALL=(ALL)NOPASSWD:/usr/sbin/ipmi-sel
icinga ALL=(ALL)NOPASSWD:/usr/sbin/ipmi-dcmi
```

nobe80, 2023-09-23:
Hi,

yes the sudoers was the problem. It works now. i have it running on an remote agent at debian linux. 
thanks for helping!

regards

closed 2023-09-25 by gschoenberger