thomas-krenn/check_ipmi_sensor_v3 #58: Your server seems to be powered off

bklier-mpl, 2022-12-13:
Hi all,
some of our SuperMicro servers (6028TR-HTR, of course not one of SuperMicros newest systems...) don't report anything back when being queried with IPMI if the system is shut down.

The check_ipmi_sensor plugin then reports "Your server seems to be powered off. (Execution of FreeIPMI returned an empty output or only 1 header row!)" which of course is absolutely correct... but it uses the "unknown" status.

It would be nice to have the possibility to tell the plugin to report back "OK" in case of not getting any data back from IPMI 'cause sometimes systems being shut down are absolutely OK for reasons of, for example, power saving concerns.

encbladexp, 2023-03-25:
Reporting Back "OK" in a State which should be "UNKNOWN" is a bad pattern. If you monitor a system which is down, use a Downtime in the Monitoring (could be scheduled).

gschoenberger, 2023-03-27:
Full ACK!
Closing this as a Downtime should be used...

closed 2023-03-27 by gschoenberger

encbladexp, 2023-04-21:
I would recommend a dedicated check for this scenario, that's what we did here: The check reached out to the vCenter to verify if a node should be down. This information could either be used to schedule a downtime, or to suppress failed states of other checks.